google/periph #114: Add support for Windows IoT for Raspbian

maruel, 2017-03-10:
## AIs

- @maruel : Fire up a [gohci](https://github.com/periph/gohci) worker running [Windows IoT](https://developer.microsoft.com/windows/iot)
- someone: Do the actual implementation, alike [host/sysfs](https://periph.io/x/periph/host/sysfs) but likely named host/winiot.

deadprogram, 2017-04-19:
@maruel do you have a link to any docs for the implementation on host/winiot? So far, all I have found is info for Windows Universal API's that all require managed (aka .NET CLR) implementations.

maruel, 2017-04-19:
I had filed the issue but hadn't seriously investigated yet. I looked at bit more now.

So "C++" is supported but it is indeed the dotnet dialect, not real C++. I was likely a bit too optimist here. An option is to create a translation DLL that exposes the subset of the needed APIs as a C API, then import this directly without using cgo. This is in the realm of the possible[1] but seems like fighting against the tide a lot with 3 layers of abstractions in the way. I was hoping more for a more direct approach like a Win32 inspired API.

[1] pending not passing afoul Universal App limitations

A good sign is that Win32 API is indeed leaking through the dotnet API, for example [GpioChangeReader](https://docs.microsoft.com/en-us/uwp/api/windows.devices.gpio.gpiochangereader#Windows_Devices_Gpio_GpioChangeReader__ctor_) can raise `HRESULT_FROM_WIN32` :)  Now if I could get access to the sources of these classes, I could get away from the whole dotnet subsystem and call into the kernel directly.

A good sign is that there is officially support for [memory mapped GPIO](https://developer.microsoft.com/en-us/windows/iot/docs/lightningproviders) but the [performance page](https://developer.microsoft.com/en-us/windows/iot/docs/LightningPerformance) seems to imply it is somewhat slow, that said I need to write a proper gpioperf to have apples-to-apples numbers to compare on Raspbian.

---

I searched a bit on https://github.com/dotnet and https://github.com/Microsoft but failed to find anything relevant. But then I found https://github.com/ms-iot/lightning/blob/develop/source/DmapSupport.h and it seems to be what I wanted! So basically open the hardcoded paths, then leverage the code paths guarded with `WINAPI_FAMILY_PARTITION(WINAPI_PARTITION_DESKTOP)` in https://github.com/ms-iot/lightning/blob/develop/source/DmapSupport.cpp#L254, basically it boils down to a standard `CreateFile()` + `DeviceIoControl()`.

**Now** we're talking! So the pseudo code would look like:

```
func GetBaseAddr() []byte {
  h := syscall.CreateFile("\\\\.\\ACPI#INT33FC#1#{109b86ad-f53d-4b76-aa5f-821e2ddf2141}\\0", ...)
  buf := DMAP_MAPMEMORY_OUTPUT_BUFFER{}
  syscall.DeviceIoControl(h, IOCTL_DMAP_MAPMEMORY, ..., &buf, ...)
  return buf.Address
}
```

No `cgo` necessary. Pure memory mapped GPIO access. I think it'd be a good start. DMAP_MAPMEMORY_OUTPUT_BUFFER is defined in [DMap.h](https://github.com/ms-iot/lightning/blob/develop/source/DMap.h).

One thing that concerns me if we were able to leverage this is how to transfer this Win32 based executable to the device, I'd need to learn more to streamline this and hopefully integrate into Visual Studio Code or something that makes it user friendly.

maruel, 2023-09-20:
Microsoft stopped developing Windows 10 IoT Core in 2018 as per https://learn.microsoft.com/en-us/previous-versions/windows/iot-core/windows-iot-core. Closing.

closed 2023-09-20 by maruel