google/periph #127: Add support to Intel Edison

crgimenes, 2017-04-18:


maruel, 2017-04-18:
The only reason I haven't tried to investigate this platform yet is simply because it is more expensive. I need to have two, one for development and one as a gohci worker. So until I decide to shell out 200~300$USD myself _or_ I receive two in the mail, this will have to wait. :/

This platform has many hardware extensions so this will require more work to expose the headers appropriately.

crgimenes, 2017-04-19:
Sorry, I wanted to be more descriptive, but I was in the middle of a lecture when I accidentally found the package. As soon as I have time I'll see if I can implement support.

deadprogram, 2017-04-19:
The Edison has several different possible configurations, depending on which breakout board is in use. Check out the Gobot implementation for some details on how the muxing needs to be done for a few of the different options https://github.com/hybridgroup/gobot/tree/master/platforms/intel-iot/edison

maruel, 2017-04-19:
Thanks @crgimenes and @deadprogram !

Ron, I see SetBoard() has to be manually called by the user. My main concern is the lack of auto-detection. I'm very ignorant about this platform; does the extension bus has support for board self-description, like the https://github.com/raspberrypi/hats protocol?

deadprogram, 2017-04-19:
Not exactly. I just added some code here which does at last part of the auto-detection (Arduino breakout vs. others), but AFAIK there is not way to determine any more specific granularity.

https://github.com/hybridgroup/gobot/commit/8a50bb24e52612ec309485f474251edd789aea64

maruel, 2017-08-12:
Since Intel discontinued the Edison on June 19, 2017, I'll close this issue. I don't see much value in spending time into adding support since the stocks will dry up. Too bad.

I'm fine with someone spending time on this but I'd recommend having it live out of the core library anyway.

closed 2017-08-12 by maruel