google/periph #130: Driver for Bosch BMP180 (i2c temperature+pressure sensor)

fd0, 2017-04-22:
Hi,

thanks a lot for this project! I've just switched over from https://github.com/kidoman/embd and like it very much. I have a couple of BMP180 temperature and pressure sensors lying around, and I'm just implementing a driver for periph.

maruel, 2017-04-22:
Cool thanks! I just bought two on aliexpress. For the BME280, writing a smoketest was easier because I could connect one over SPI and one over I²C and compare the measured values.

For this one this would require two separate I²C or an I²C multiplexer like TCA9548A. I bought 2x TCA9548A in case this is useful.

I was thinking about leaving one of each on a gohci worker and just report if the values are "sensible", e.g. 15C < x < 30C.

fd0, 2017-04-22:
That's nice hardware, I haven't really looked around for something like this! It's a bummer most i2c devices have a hardcoded address, and in this case it's even the same as for the bme280 (0x77)...

fd0, 2017-04-22:
What's the long term plan for periph with the smoke tests? Do you plan to continue buying hardware for the drivers people implement and run them on your farm? :)

maruel, 2017-04-22:
For now as long as the devices are inexpensive, yes. There's a trade off between asking people to run gohci workers vs just buying it. My focus remains on host interfacing more than broad devices support so unless there's a deluge of contribution, it should scale fine.

Keep in mind I have a quite high bar for driver quality and unit test code coverage.

Tangentially, one thing that bothers me with the bme280 driver is the lack of good API for continuous mode. Will have to think about it more.

PR #132 (2017-04-29): Add driver for Bosch BMP180 temp/pressure sensor
This PR adds a driver for the Bosch BMP180 temperature/pressure sensor. At the moment the driver works and reads sensible data.

Closes #130
closed 2017-04-29 by maruel