google/periph #141: Support devices on the Enviro pHAT

maxekman, 2017-05-31:
The Enviro pHAH is an affordable set of sensors for environment sensing. This issue is a public todo for me when implementing the drivers for it

Devices on the HAT:
- [ ]  BMP280, temperature/pressure sensor
- [ ]  TCS3472, light and RGB colour sensor
- [ ]  LSM303D, accelerometer/magnetometer sensor
- [ ]  ADS1015, 4-channel 5v tolerant 12-bit ADC
- [ ]  Two GPIO controlled LEDs for illumination of the light sensor, not really a device

Manufacturer: https://shop.pimoroni.com/products/enviro-phat
Device info: https://pinout.xyz/pinout/enviro_phat

maruel, 2017-05-31:
I found it on [Adafruit](https://www.adafruit.com/product/3194) but it's out of stock. Pimoroni has even more reasonable shipping rates to Canada than Adafruit so I may order from there.

For now add it to experimental/ and once you have a smoke test and good unit test coverage, we can move it back into the official tree.

maruel, 2017-05-31:
I prefer per device smoketest, as it's more reusable.

UnAfraid, 2017-08-25:
We may scratch BMP280 of the list since https://github.com/google/periph/commit/252733acf606c513fa14dcff77a61fe0051b67d6

maxekman, 2017-08-27:
I'll leave it checked in the list as that device is still on the Enviro pHAT.

bezineb5, 2018-11-03:
PR #304 implemented ADS1015, so you can check it.

NeuralSpaz, 2018-12-18:
I have a discrete TCS3472 and am willing to contribute device driver.

maruel, 2023-09-20:
Ported to https://github.com/periph/devices/issues/61.

closed 2023-09-20 by maruel