google/periph #144: Support for C.H.I.P Pro

VALERE91, 2017-06-15:
Hi,

I have several C.H.I.P Pro and it's a really nice and affordable platform. The Periph website says that this platform will be supported soon. Have you any idea when this support will be achieve?

Thanks

maruel, 2017-06-15:
Hi!

I bought the dev kit so I have two at home but hadn't taken the time to do this yet. The AIs are:

- Add code for auto detection in [chip.go](https://github.com/google/periph/blob/master/host/chip/chip.go). I think [rpi.go](https://github.com/google/periph/blob/master/host/rpi/rpi.go#L123) can serve as a good example how to support multiple board types.
- Tweak [chipsmoketest.go](https://github.com/google/periph/blob/master/host/chip/chipsmoketest/chipsmoketest.go).
- Add one CHIP Pro to the gohci workers (I can do that).
- Update the [website](https://github.com/periph/website).

If you have time to work on this, I'll happily review, otherwise I can raise the priority of this, now knowing that there are interested users.

tve, 2017-06-16:
Dumb question: other than the headers being different, is there anything that doesn't work?

VALERE91, 2017-06-19:
Hi!

Sorry for this delay before my answer I was pretty busy. 

@maruel I  don't think I will have time for this before a month or two so if you can raise the priority it will be very nice.

Thanks

mattetti, 2017-08-19:
I also ordered a dev kit which should ship in September. @VALERE91 did you have a chance to look into what doesn't work?

VALERE91, 2017-08-19:
Sadly no I'm on a big rush on other projects. I will try to raise this priority as soon as possible since I will need it for a project in September.

maruel, 2018-01-03:
I revisited this. NTC officially deprecated their Debian image and wants everyone to use their GadgetOS buildroot derivative that is read-only except for running docker containers. It made me think of [Google's Container-Optimized OS](https://cloud.google.com/container-optimized-os/) which I have familiarity with.

Their Debian image is still on Jessie. I just tested it and it fills up the available space quite fast. I'm at 212Mb free with only ssh installed, ~230Mb of usage.

I haven't tried their GadgetOS yet but NTC reports a tad above 60Mb of flash use. The main downside is that GadgetOS looks "alpha quality" at best and is developed by a very small team.

Even with the downside, my recommendation is to support CHIPPro only on GadgetOS. This means significant changes to this FR.

Since Go can be statically compiled, there is no need to map a user space environment like Alpine, like it is done in this python based container: https://github.com/xtacocorex/gadget-chip-io. This means that it would be possible to create a standalone environment that would work great by mapping /sys and /dev/mem. The net result is that there is value in having a base Dockerfile that describes this baseline. That said, this Dockerfile doesn't belong inside https://github.com/google/periph. I'm fine with putting this into https://github.com/periph/extra. This requires also significant documentation and testing coverage.

Refs:

- Doc: https://docs.getchip.com/gadget.html
- Code: https://github.com/NextThingCo/gadget-os
- CLI tool: https://github.com/NextThingCo/gadgetcli

maruel, 2018-08-08:
Sadly, NTC closed its door, so while I'll gladly accept changes related to this, I don't feel the need to keep the issue open.

closed 2018-08-08 by maruel