google/periph #146: unneeded external dependency?

mattetti, 2017-08-11:
I noticed that github.com/kr/pretty is a dependency of the `allwinner` package (see dma.go). Is that really necessary?

maruel, 2017-08-11:
Ah no it's an oversight that I pushed it, as used it while debugging but it was not meant to be upstream. But I just realized it's not the only one.

## AI

- Remove the dependencies and push that out.
- Add a formal test in `.travis.yml` so there's no regression, something I wanted to do for a long time but never kicked myself to do. The following command works great to get the list, just need to have it exit 1 when it's not empty:
```
go list -f '{{ join .Imports "\n" }}' periph.io/x/periph/... | sort | uniq | grep -v '^periph.io/x/periph' | xargs go list -f '{{ if not .Standard }}{{ .ImportPath }}{{end}}
```

It currently prints out:

```
src/periph.io/x/periph🐟 go list -f '{{ join .Imports "\n" }}' periph.io/x/periph/... | sort | uniq | grep -v '^periph.io/x/periph' | xargs go list -f '{{ if not .Standard }}{{ .ImportPath }}{{end}}'
github.com/kr/pretty
golang.org/x/image/font
golang.org/x/image/font/basicfont
golang.org/x/image/math/fixed
```

The culprits are:
- cmd/ssd1306/main.go
- devices/ssd1306/ssd1306_test.go
- devices/ssd1306/ssd1306smoketest/ssd1306smoketest.go
- host/allwinner/dma.go
- host/allwinner/spi.go

The ssd1306 one can remove the font support, and I can migrate it to periph/extra. The allwinner one is just removing the code and printing it without formatting.

maruel, 2017-08-11:
`github.com/kr/pretty` is now fixed, `golang.org/x/image/...` remains before adding the check.

commit cef73a01 (2017-08-11): Remove reference to github.com/kr/pretty

It was pushed by error into work-in-progress code.

Fixes #146
commit 19ff685d (2017-08-12): travis: Add check to error out on any external imports

No regression can silently creep in.

Fixes #146
commit 543ad8e4 (2017-08-12): travis: Add check to error out on any external imports

No regression can silently creep in.

Fixes #146
commit 7b61e537 (2017-08-12): travis: Add check to error out on any external imports

No regression can silently creep in anymore.

Update Go used in travis to 1.8.x.

Fixes #146
closed 2017-08-12 by maruel