google/periph #155: Support for BMP280

UnAfraid, 2017-08-22:
Hello,

Can you please add support for the BMP280 sensor, i see that BME280 is supported which is basically the same (With the lack of humidity) but both sensors are backward and forward compatible, just the chip id is different 0x58 for BMP280 and 0x60 for BME280.

Thanks.

maruel, 2017-08-22:
Ok it seems quite straightforward. I've ordered two on aliexpress for now.

The main annoyance is the 2s/4s standby but since I've unexported Standby in af3fbceec62dc44086c6731d9f181f33dc92a26c, this simplifies hiding this discrepancy.

I'm wondering what to do with the driver name though, as it's not really worth having a completely separate package, since the actual device can be easily detected. I guess `bmx280` would be fine, albeit slightly confusing. :/

Another option is to create a directory `bosch`, then put the devices in there as subdirectories so it's easier to share the code via an `internal` package.

I'm open to other ideas too.

maruel, 2017-08-22:
I've made a commit showing how it could me done. I think it's feasible, the main annoyance is the user needs to know "more" about the right I²C address to use.

quinte17, 2017-08-22:
maybe device-creation needs a bit of a rework?
what about something like
func NewBMP180() (clever_device_object)
the only problem i see here, is how to get the difference between spi and i2c.
NewBMP180I2C
NewBMP180SPI
seems a bit redundant?

UnAfraid, 2017-08-22:
Thanks for the quick reply i really didn't wanted to fork the repo just for a single byte change i really like the idea of not duplicating the driver but naming it bmx280, great idea!

maruel, 2017-08-22:
@quinte17 the function NewI2C() handles all of bmp180/bme280/bmp280 seamlessly as per the WIP code I pushed. The only challenge is the address: I'm thinking it would be split off as a separate argument in NewI2C, so it couldn't awkwardly be passed to NewSPI due to its presence in Opts{}.

maruel, 2017-08-22:
FTR, I'm planning a v2 release really soon, so getting this in before the v2 would be a good idea since it's totally a breaking change.

quinte17, 2017-08-22:
With the options...
the only thing I can imagine is to use different types or different interfaces.
or just live with it and just write it into the docs that address wont have any affects when using spi.

maruel, 2017-08-22:
I changed NewI2C() to have an additional argument. It's not that bad (3 args now) and makes the code simpler.

maruel, 2017-08-23:
@UnAfraid can test 2ba2860 locally? You can get with it:

```
git fetch && git checkout origin/bosch
```

Even better would be to connect one BMP280 device over I²C, one over SPI then run:
```
periph-smoketest bmx280
```
and confirm that both readings are close, which means both protocols work correctly. This would greatly increase the confidence level for the whole thing working.

Thanks

UnAfraid, 2017-08-23:
Sure gonna try it.

UnAfraid, 2017-08-23:
I couldn't get nor bme nor bmp to work under SPI, on i2c it works fine tho.
It says
```chip@chip4:~/go/src/periph.io/x/periph/cmd/bmxx80$ sudo ./bmxx80 -spi /dev/spidev32766.0 -v
19:34:12.601514   CLK : INVALID   
19:34:12.602751   MOSI: INVALID   
19:34:12.603339   MISO: INVALID   
19:34:12.603854   CS  : INVALID```

maruel, 2017-08-23:
The SPI driver on the CHIP is, I'd say, "_subpar_". I'm only testing SPI on a Raspberry Pi. But that's fine I'll confirm first I didn't break anything (likely tomorrow or the day after) with my 2 BME280 and that should be good enough.

UnAfraid, 2017-08-23:
I have Rasberry PI Zero around, i can try with it.

UnAfraid, 2017-08-24:
Here's smoketest output from 2 BMPs 
```bash
pi@pi-zerow:~/go/src/periph.io/x/periph/cmd/periph-smoketest $ ./periph-smoketest bmx280 -i2c 1 -spi /dev/spidev0.0
BMP280{I2C1(118)} : 25.830°C 100.825KPa   0.00%rH
BMP280{SPI0.0}    : 26.060°C 100.814KPa   0.00%rH
Delta             :  0.230°C   0.011KPa   0.00%rH
```

maruel, 2017-08-25:
Thanks a lot for the verification. I added the flag to the smoke test.

commit 2f811d03 (2017-08-25): bmxx80: Merge bmp180 and bme280 and add support for bmp280

- cmd/bmxx80 now can read any of these 3 devices.
- Rewrite the driver to support all 3 devices. More devices of the family (like
  the bme680) could be easily added.
- Add -ia to bmx280smoketest

Fix #155
PR #156 (2017-08-25): bmxx80: Merge bmp180 and bme280 and add support for bmp280
- cmd/bmxx80 now can read any of these 3 devices.
- Rewrite the driver to support all 3 devices. More devices of the family (like
  the bme680) could be easily added.
- Add -ia to bmx280smoketest

Fix #155
commit 252733ac (2017-08-25): bmxx80: Merge bmp180 and bme280 and add support for bmp280 (#156)

- cmd/bmxx80 now can read any of these 3 devices.
- Rewrite the driver to support all 3 devices. More devices of the family (like
  the bme680) could be easily added.
- Add -ia to bmx280smoketest

Fix #155
closed 2017-08-25 by maruel