google/periph #163: Config options for strong pullup on the ds248?

justonia, 2017-09-23:
In the setup process for the ds248 there is no ability to set the config bit to use a strong pullup:

```
d.confReg = 0xe1 // standard-speed, no strong pullup, no powerdown, active pull-up
if opts.PassivePullup {
	d.confReg ^= 0x11
}
```

Is there a reason this does not exist or could I implement it as an exported option and make a PR?

tve, 2017-09-23:
The strong pull-up is not something that gets configured but something that gets turned on when it is needed. From the datasheet (ds2483 p.8):
> The SPU bit is used to activate the strong pullup function
prior to a 1-Wire Write Byte or 1-Wire Single Bit
command. Strong pullup is commonly used with 1-Wire
EEPROM devices when copying scratchpad data to the
main memory or when performing a SHA computation
and with parasitically powered temperature sensors or
A/D converters.

See https://github.com/google/periph/blob/master/devices/ds248x/dev.go#L51-L57

justonia, 2017-09-23:
Ah ok, I think I understand. Then is the issue in the DS18B20 code and its
inability to let me specify what pull-up type is sent to the DS248 one-wire
interface?

I’m trying to power that temperature sensor via parasitic power and without
a strong pull-up it will spew errors on temperature reads until, I assume,
it’s internal capacitor is full enough to do a conversion.

tve, 2017-09-23:
The DS18B20 driver uses the strong pull-up: https://github.com/google/periph/blob/master/devices/ds18b20/ds18b20.go#L82-L84 and https://github.com/google/periph/blob/master/devices/ds18b20/ds18b20.go#L108
I have 14 parasitically powered DS18B20 on one DS2483 and they have been running fine with a read-out every second for over a year.

maruel, 2017-12-09:
I'm not familiar with this device, is there any action item left here?

maruel, 2018-07-17:
Closing due to inactivity. Please reopen if there's still an action item.

closed 2018-07-17 by maruel