google/periph #166: ssd1306 wrong resolution on 128x32 OLED

mikegleasonjr, 2017-10-23:
Using this product:

Adafruit PiOLED - 128x32 Monochrome OLED Add-on for Raspberry Pi
https://www.adafruit.com/product/3527

The image seems wrong using the `ssd1306` cmd provided in the package:

```
./ssd1306 -h 32
```

Results in:

[image]

It only writes to the second half of the screen, with tiny text cut in half.

```
# even though the display has a height of 32, I tried 64 to see
./ssd1306 -h 64
```
Results in:

[image]

The display is working fine with their official python library: https://github.com/adafruit/Adafruit_Python_SSD1306/blob/master/Adafruit_SSD1306/SSD1306.py

The **Adafruit 128x64 OLED Bonnet for Raspberry Pi**  is working fine with periph (https://www.adafruit.com/product/3531): 

[image]

maruel, 2017-10-23:
Interesting. Can you try changing this line:
https://github.com/google/periph/blob/master/devices/ssd1306/ssd1306.go#L207
to `0xDA, 0x02,` ?

mikegleasonjr, 2017-10-23:
@maruel nope, same problem.

I don't know if it is relevant/interesting or not but... if I try with `-h 32` on the 128x64 display, it displays on the bottom half of the screen (like 1st screenshot, but on the 128x64 device)

mikegleasonjr, 2017-11-16:
@maruel I picked up the device again and found the problem this time.

I looked at how the device was initialized with the adafruit python library (https://github.com/adafruit/Adafruit_Python_SSD1306/blob/master/Adafruit_SSD1306/SSD1306.py#L309). I found a mismatch here https://github.com/google/periph/blob/master/devices/ssd1306/ssd1306.go#L207.

So I replaced...

```
0xDA, 0x12, // Set COM pins hardware configuration; see page 40
```

with...

```
0xDA, 0x02, // Set COM pins hardware configuration; see page 40
```

And now the display is working properly:

[image]

There is no public interface to send commands directly to the display, what do you propose to make something flexible so I can make it work?

mikegleasonjr, 2017-11-17:
@maruel I feel silly, that was your proposed diagnosis the first time!

mikegleasonjr, 2017-11-17:
I found out how to do it:

```
if err = c.Tx(0x3C, []byte{
    0x00, // I²C transaction has stream of command bytes
    0xDA, // set comm pins
    0x02, // 2
}, nil); err != nil {
    return nil, err
}
```

I am going to close it now :)

maruel, 2017-11-20:
No worries. :) Reopening because this is a bug.

So clearly the COM pins configuration needs to either have more conditions or be made configurable. I think we'll need the later as this really depends on the hardware configuration. :/ Expanding on the doc which isn't that clear;

Bits
- 5 enables top and bottom panel swap
- 4 enables alternative each side instead of using two subpannels

This is coupled with C0/C8:
- C0 means bottom-top
- C8 means top-bottom

It is used to rotate the display.

So we'll need an enum to clarify the above in a way that keeps the rotated functionality working (which I'd prefer if possible).

maruel, 2017-12-09:
FTR, I still want to fix this, I just need to figure out the best way to expose the bit.

maruel, 2017-12-12:
I'm starting to think about v3, this should be fixed in time for this release, which will likely be in a few months.

mikegleasonjr, 2017-12-12:
@maruel thanks! in the meantime I have a workaround so that's great

closed 2018-09-03 by maruel

mikegleasonjr, 2018-09-03:
Thank you very much!

PR #282 (2018-09-03): ssd1306: add options to customize the hardware pin layout
Fixes #166
commit 07bdd3d5 (2018-09-03): ssd1306: add options to customize the hardware pin layout (#282)

Fixes #166