google/periph #167: Support for Industrial I/O (iio)?

ubergesundheit, 2017-10-31:
Do you plan to add support for interfacing with devices through the Industrial I/O linux subsystem? It features many device drivers already baked into the linux kernel (see [here](https://github.com/torvalds/linux/tree/master/drivers/iio) for a list). 

I would very much like the idea of using go for reading values from sensors connected through iio.

maruel, 2017-10-31:
I wasn't aware so there's no plan. :) If RPi's can get recent drivers (as many seem to have been touched in the past year), it would be interesting.

I looked on a RPi3, ODROID and x64 host running Ubuntu 16.04 and failed to see `/sys/bus/iio`, as documented athttps://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/ABI/testing/sysfs-bus-iio. It seems like a device tree overlay is needed, which is non-trivial for users.

If this ever to work out, the plan would be to create a package periph.io/x/periph/host/iio`, have it query with whatever interface and populate what is available here.

maruel, 2017-11-08:
`host/fs` exists to enable unit testing so it's a good idea to use it for this reason, but it's not an hard requirement.

You could start in `experimental/host/iio` and start to work iteratively from there, and then once it's deemed stable enough we can move it back as `host/iio`. So if you want to commit to implement at least code to have a functional baseline, I'm fine with you start contributing into experimental. You may also want to do an initial prototype on a private branch if you want, just to weed out the very initial key design elements. It may be challenging to design especially as I think you may need to create a "registry of devices found". It's as you prefer.

As part of this task, another important point will be to document how to enable it on https://github.com/periph/website, not sure where exactly yet.

Note that this code _could_ live in a separate repository just fine too. Periph is intentionally designed to allow that. It depends on how much velocity you want to have when pushing code.

maruel, 2023-09-20:
Ported to https://github.com/periph/host/issues/45.

closed 2023-09-20 by maruel