google/periph #168: Getting PWM to work with Raspberyyi Pi Zero W

mikegleasonjr, 2017-12-03:
Hi,

Whenever I use `pi-blaster` to output a `pwm` signal on a pin, it works correctly (the led turns on):

```
sudo ./pi-blaster -D -g 18
...
echo "18=0.75" > /dev/pi-blaster
```
But when I try with `periph`, it doesn't work (I have no errors, but the pin doesn't output any current, the LED stays off):

```
package main

import (
	"fmt"
	"os"
	"os/signal"
	"time"

	"periph.io/x/periph"
	"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/host/rpi"
)

func main() {
	periph.Init()
	p := rpi.P1_12 // GPIO18
	pwm := p.(gpio.PinPWM)

	err := pwm.PWM(gpio.Duty(50535), 100*time.Microsecond)
	if err != nil {
		fmt.Fprint(os.Stderr, err)
		return
	}

	c := make(chan os.Signal, 1)
	signal.Notify(c, os.Interrupt)
	<-c
}
```

[image]

maruel, 2017-12-03:
Did it print an error message? To work, you have to run it as root. Also I never tested having both loaded, as pi-blaster will also try to setup the PWM clock source.

There's also known bugs that the clock is not canceled correctly yet; https://periph.io/x/periph/host/bcm283x#pkg-note-bug

mikegleasonjr, 2017-12-03:
Hi !

* No error messages (`err == nil`)
* I noticed that `pi-blaster` and my own code can conflict, I tried both separately after clean reboots. Whenever I started `pi-blaster`, I started it as not deamon (`-D`) so that I could stop it.

I am using the latest `Stretch` (end of november) with a clean install.

maruel, 2017-12-04:
Just a note that pi-blaster plays with the same CPU registers and the state modification is not bound by process lifetime. Sometimes resetting the state requires really cutting the power.

That said the implementation is still known to be buggy but should have worked. Will double check in the next days.

simokawa, 2017-12-04:
I also found current hardware PWM does not work. I found a few problems.
1. PWM address map is broken. For example,  pwmMemory.rng1 is offset by 4 bytes, we need a padding before it.
2. Current configuration of dat1 looks strange.
I will send a pull request for 1.

I am using Pi B+.
Thanks,

mikegleasonjr, 2017-12-06:
Still not working with a clean power-cycle and #169 merged

maruel, 2017-12-06:
Thanks for the report. Next window I have available to investigate at the earliest is tomorrow evening. Will report once I double checked what I messed up.

simokawa, 2017-12-07:
Please try my pwm2 branch
https://github.com/simokawa/periph/tree/pwm2
or the patch
https://github.com/simokawa/periph/commit/9d5eacd5a5db312b9983ff2174aaf9a35e8e8e45

It also modifies rng/dat settings + verbose debugging and worked for me.

maruel, 2017-12-07:
Thanks! Will try to look at it tonight.

mikegleasonjr, 2017-12-07:
@maruel @simokawa I can confirm it is working on the `pwm2` branch:

```
package main

import (
	"os"
	"os/signal"
	"time"

	"periph.io/x/periph"
	"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/host/rpi"
)

func main() {
	periph.Init()
	p := rpi.P1_12 // GPIO18
	pwm := p.(gpio.PinPWM)

	c := make(chan os.Signal, 1)
	signal.Notify(c, os.Interrupt)

	t := time.NewTicker(100 * time.Millisecond)
	defer t.Stop()

	i := gpio.Duty(0)
	step := gpio.Duty(2000)
	for {
		select {
		case <-c:
			return
		case <-t.C:
			pwm.PWM(i, 100*time.Microsecond)
			i += step
			if i > gpio.DutyMax || i < 0 {
				step = -step
				i += step
			}
		}
	}
}
```

maruel, 2017-12-09:
Ok finally had time to confirm on the oscilloscope it works great, can you clean up the branch in preparation for merging?

maruel, 2017-12-09:
That's some success :)

[image]

simokawa, 2017-12-09:
Nice ;-)

I did more tests and find out we need to clear unnecessary ctl bits otherwise GPIO12 did not work after reboot.

I removed debug outputs and made a pull request.

closed 2017-12-09 by maruel

mikegleasonjr, 2017-12-09:
thanks guys!

commit 332ac51f (2017-12-09): Make hardware PWM work. (#171)

* Fix PWM Address Map: add missing paddings.
* Set PWM rng/dat correctly.
* Clear ctl bits before set.

Fixes #168