google/periph #170: sonoff devices

gedw99, 2017-12-07:
https://github.com/arendst/Sonoff-Tasmota

these devices are 5 bucks each and allow OTA update over wifi once initially flashed.

the range of hardware is high. power outlets, light switches, LEDS, and a fair few other things
they all have a wifi chip and so make sit easy to use with mqtt etc.

wondering if anyone else sees potential with these.
to me they look very good.

maruel, 2017-12-08:
Please only file bug reports or feature requests on the issue tracker. General discussion should go on the slack channel.

This project is GPL3 so we're not going to touch it even with a 10 foot pole.

closed 2017-12-08 by maruel

maruel, 2017-12-08:
My reply was a bit harsh so expanding to explain the rationale;

periph.io is licensed as Apache v2.0 because the goal is to make it reusable in commercial projects. As such, it is incompatible with GPL v3 projects. Here's a short primer by the Apache foundation; https://www.apache.org/licenses/GPL-compatibility.html

The ESP8266 is a great device, for example I played with it in https://github.com/maruel/bolide-esp8266. That said, it explicitly cannot run Go programs. As such, this kind of device cannot be handled directly. Also, periph.io is specifically designed as a zero external dependency project focused on low level support. You can read more about this at https://periph.io/project/goals/

Thanks.

gedw99, 2017-12-08:
no probs.

i had a look at the https://github.com/maruel/bolide-esp8266 project.
What i wanted to do was use golang to update the chip OTA.
This guy cracked it and explains how here:
http://blog.nanl.de/2017/05/sonota-flashing-itead-sonoff-devices-via-original-ota-mechanism/
his code: https://github.com/mirko/SonOTA

---

So i noticed that that you have a really good update tool (https://github.com/periph/bootstrap), and was talking about trying to adapt it to also update the ESP8266