google/periph #189: gpiostream: Remove Bits and BitStream.

maruel, 2017-12-15:
They are superseded by the LSB and MSB variants.

simokawa, 2017-12-20:
MSB may not be so efficient as expected in little endian system. One possible workaround is using []uint32 instead of []byte.
Here is an example, suppose a user want to send bytes, 0x1 0x2 0x3 0x4 in this order and MSB first:
s.Bits[0] = 0x01
s.Bits[1] = 0x02
s.Bits[2] = 0x03
s.Bits[3] = 0x04

When CPU or DMA read these and write to FIFO as 32bit word, these are read as 0x04030201 because the system is little endian. PCM or PWM will send this 32 bit word from MSB and this is not expected by the user.

Possible solutions are:
- use []uint32 for Bits
- swap bytes when writing to DMA buffer or FIFO (inefficient)
- 8 bit transfer (inefficient)

@maruel 
- Assuming 32bit fifo is too hardware specific?
- Could system be big-endian?

maruel, 2017-12-20:
- Go requires a >=32 bits processor.
- The chances of running on a big endian processor is very, very close to zero but I wouldn't rule it out completely. Obviously should be optimized for little endian processor.

Was probably an error to do the bit streams as 8 bits vs 32 bits, but there's occasionally value. As a matter of fact, https://periph.io/x/periph/host/pmem#Slice could be of help here, as we could generalize it (?)

simokawa, 2017-12-23:
I decided to use the second option, byte swapping. I guess it should work in big endian system although it also depends on bus architecture. I heard ARM can switch to big endian, so theoretically bcm283x can be big endian?  

BTW, I am also a bit frustrated with the units.
- period vs frequency: we cannot use arbitrary clock frequency, rounding error during conversion between period and frequency makes it hard to specify intended frequency.
- duty: DutyMax is an odd number, we cannot specify 50% duty precisely.

commit 4b3fb21b (2018-03-26): gpiostream: rename LSB->LSBF and MSB->MSBF

The first part is normally implied but for newcomers is unclear. Having the work
in the alias will be a good reminder.

Remove Bits and BitStream.

Fixes #189
commit d457e36c (2018-03-26): gpiostream: rename LSB->LSBF and MSB->MSBF

The first part is normally implied but for newcomers is unclear. Having the work
in the alias will be a good reminder.

Remove Bits and BitStream.

Fixes #189
commit efe8ccda (2018-03-26): gpiostream: rename LSB->LSBF and MSB->MSBF

The first part is normally implied but for newcomers is unclear. Having the work
in the alias will be a good reminder.

Remove Bits and BitStream.

Fixes #189
commit 086d21ca (2018-04-06): gpiostream: rename LSB->LSBF and MSB->MSBF

The first part is normally implied but for newcomers is unclear. Having the work
in the alias will be a good reminder.

Remove Bits and BitStream.

Fixes #189
closed 2018-05-28 by maruel