google/periph #201: beagle: implement header tables for beaglebone/beagleboard board family

maruel, 2017-12-22:
It will be useful coupled with am335x support (issue #199) and PRU-ICSS support (issue #200).

https://beagleboard.org/

bodokaiser, 2018-04-11:
For the raspberry pi you [set up the pin layout](https://github.com/google/periph/blob/master/host/rpi/rpi.go#L50) with [bcm283x.GPIO](https://github.com/google/periph/blob/master/host/bcm283x/gpio.go#L25).

For the Beaglebone (Black) the analogue would be the am335x?

maruel, 2018-04-11:
Yes, there's scaffolding at http://periph.io/x/periph/experimental/host/am335x and http://periph.io/x/periph/experimental/host/beagle.

In practice a full am33x implementation is not initially needed, only detecting the presence. But even so, I think there's beaglebones on different processors (?) so detecting the board is what is important here, and registering the right header.

bodokaiser, 2018-04-11:
I only used the Beaglebone Black so far. Detection is done rudimentary by checking sysfs files?

How do you plan to support pin configuration (as we discussed on slack)? We could try to mimic the `config-pin` tool which I think just interfaces sysfs to find the current pin configuration, however this would make things more complicated. Alternatively we could just allow the user to overwrite specific pins?

maruel, 2018-04-11:
Detection should be done via `/proc/device-tree`.

Let's aim to have the header definition first, and add magicness later.

maruel, 2018-07-16:
Done (enough for now)

closed 2018-07-16 by maruel