google/periph #206: Stream.Duration() bug

simokawa, 2017-12-23:
If I understand correctly, Stream.Resolution() is duration for a single bit and Duration() should return for all **bits***.
Current implementation uses the count of **bytes** instead of **bits**, this looks wrong to me.
https://github.com/google/periph/blob/master/conn/gpio/gpiostream/gpiostream.go#L68
It should be
`return b.Res * time.Duration(len(b.Bits)) * 8`
Yeah, it is confusing that len(b.Bits) returns number of **bytes**. 

There may be some code assuming this bug.

maruel, 2017-12-23:
Yep :(

simokawa, 2017-12-24:
Assuming number of bits to be 8*len(b.Bit) may work for most cases but we may want to specify number of bits explicitly in the struct. For example, gpiostream.Program{} could be a loop of 3bit-stream.

maruel, 2023-09-20:
Ported to https://github.com/periph/conn/issues/34.

closed 2023-09-20 by maruel