google/periph #219: Initialize only I2C stack

jdevelop, 2018-02-25:
Is there any easy way to initialize just I2C bus, not SPI/GPIO/etc? I have some custom code with bit banging over SPI, and it goes down to the conflict because of the resource usage.

maruel, 2018-02-25:
A few questions to clarify:
- Do you mean https://periph.io/x/periph/host/sysfs specifically?
- What platform, a Raspberry Pi running Raspbian or something else?
  - In this case, is there any reason why you can't use `raspi-config` to disable SPI?
- What do you mean exactly by "resource usage"?
- Do you need to use the exact same SPI pins or any pin would do?
- [driverSPI.Init()](https://github.com/google/periph/blob/master/host/sysfs/spi.go#L506) specifically doesn't open the SPI devices, so I'm surprised there would be any interference there.

jdevelop, 2018-02-25:
@maruel I just rebooted the PI and now everything works fine, I assume that there was something in the way. Closing this as not an issue.

closed 2018-02-25 by jdevelop