google/periph #226: gpio.PWM() Interfearing with gpio.Out()

Ivaan, 2018-03-28:
At it simplest a call to
``bcm283x.GPIO25.PWM(dutyCycle, period) //rpi.P1_22``
causes a subsequent call to
``bcm283x.GPIO27.Out(gpio.High) //rpi.P1_13``
to not do-the-right-thing.
**Observed behaviour:** physical pin 13 outputs a random square wave
**Expected behaviour:** physical pin 13 should output a high signal

	package main

	import (
		"bufio"
		"fmt"
		"os"
		"time"

		"periph.io/x/periph/conn/gpio"
		"periph.io/x/periph/host"
		"periph.io/x/periph/host/bcm283x"
	)

	func main() {
		reader := bufio.NewReader(os.Stdin)
		host.Init()

		fmt.Println("one")
		fmt.Println(" - both pin 22 and 13 low or the state they were in at the end of the previous run")
		reader.ReadRune()

		dutyCycle := gpio.Duty(gpio.DutyHalf)
		period := time.Millisecond * 20.0
		bcm283x.GPIO25.PWM(dutyCycle, period) //rpi.P1_22

		fmt.Println("two")
		fmt.Println(" - pin 22 50% dutyCycle 20ms period")
		fmt.Println(" - pin 13 low first run after reboot, random square wave if not first run")
		reader.ReadRune()

		bcm283x.GPIO27.Out(gpio.High) //rpi.P1_13

		fmt.Println("three")
		fmt.Println(" - pin 22 50% dutyCycle 20ms period")
		fmt.Println(" - pin 13 random square wave")
		reader.ReadRune()

		bcm283x.GPIO25.Halt()

		fmt.Println("four")
		fmt.Println(" - both pin 22 and 13 low or high")
	}

I've observed this happening to more than just pin13 though only after setting an output. I've not done an exhaustive search. This random square wave behaviour seems to persist multiple runs and clears on reboot.

maruel, 2018-03-28:
Thanks! Paging @simokawa who authored this in case he has an idea what's happening.

simokawa, 2018-03-28:
Thanks for the report! Unfortunately, I may not be able to test until next
week end.

DMA writes to 32bit register which controls multiple pins, so it may
transferring wrong bit masks.
However, It is controlled by two(high/low) simple control block(cb) and I
have no idea why pin 13 output random bits while pin 22 is periodic.
https://github.com/google/periph/blob/master/host/bcm283x/dma.go#L766

Hidetoshi

simokawa, 2018-04-18:
I tried to reproduce the issue today but could not.
I run the sample program above multiple times but GPIO27 keeps high state, no random wave.
Can you trigger random waves just by the second run of the program?

My random guess is interfering from other applications. In my experience, pi-blaster uses DMA and interfered with periph.

maruel, 2018-04-18:
@Ivaan One quick question: what distro did you use? I only test on Raspbian Stretch Lite; e.g. without GUI. This may be important.

simokawa, 2018-04-20:
Kernel may have something to do with because we are stealing resource from it :-)

https://periph.io/platform/raspberrypi/ suggest to comment out "dtparam=audio=on" in /boot/config.txt
@maruel do you know how audio=on affects us?

maruel, 2018-07-17:
Side note: I think we should disable PWM support when we detect X is running.

gregzuro, 2018-08-09:
I've just came across something similar to this.

If I do some PWM stuff on P1_22, I see the same waveform on P1_8 as well.  The relay on that latter pin wasn't too happy. :(

bezineb5, 2020-04-16:
I have similar issues: I'm trying to control 2 servos on GPIO23 and GPIO24.
If I use only one of them, it works perfectly. No jittering, and the servo can travel the full range.
If I use both at the same time, GPIO24 seems to be working fine, but GPIO23 jitters and is limited in range.
I guess that the signal on GPIO23 is not clean, but I don't have an oscilloscope to check. Any idea?

bezineb5, 2020-04-16:
I think I found a solution, at least for my problem: it seems that the txLen parameter of the controlBlock structure leads to that instability. By "exploding" the control blocks (ie, there are as many controlBlocks as there are cycles in the period - all with a fixed length), everything works fine, as with pi-blaster.
I can do a PR if it's seen as not too ugly...

commit 844755e8 (2020-04-18): bcm283x: Fixes #226: unrolling DMA controlBlocks (#440)

Fix for issue #226: the DMA controlBlocks are unrolled and the parameter txLen is set to a fixed value.

Moved control block size to a constant.
closed 2020-04-18 by maruel