google/periph #238: spi half duplex fails

bodokaiser, 2018-04-25:
When passing `spi.HalfDuplex` as parameter to `spidev.Connect(...)` I get

>sysfs-spi: setting mode Mode3|HalfDuplex failed: invalid argument

I was able to track the error down to this line:

https://github.com/google/periph/blob/a72045d66f9be001cc8da21c9f67ef35d54f1675/host/sysfs/spi.go#L299

Any ideas what goes wrong over here?

maruel, 2018-04-25:
Maybe the driver didn't want it? I haven't tested this flag at all. :(

bodokaiser, 2018-04-25:
I wonder about the error code? Further the [kernel docs][1] actually say that half duplex is enabled per default.

Any suggestions how I could test this detached from go / periph.io? What docs did you use for spi maybe I can write a c file to test.

1: https://www.kernel.org/doc/Documentation/spi/spidev

maruel, 2018-04-25:
I looked at `/usr/include/linux/spi/spidev.h` on my workstation and built a small C program to confirm the values.

maruel, 2023-09-20:
I'll close this issue since it is quite old. If this persists please reopen at https://github.com/periph/host. Thanks!

closed 2023-09-20 by maruel