google/periph #256: bmxx80: Humidity is 10x smaller than it should be

zbikmarc, 2018-07-16:
Humidity stored in `physic.Env` is 10x smaller than it should be.

Output shown by example code for bmxx80:
`24.370°C  97.648kPa    5.3%rH`

Example output from Python library for bme280:
`humidity=53.29 % rH`

Example output from c library for bme280
`"humidity":53.01`

maruel, 2018-07-16:
Which peripheral version are you using?

maruel, 2018-07-16:
err "periph"

zbikmarc, 2018-07-16:
that was fast... 
I am using 'master'. Sorry for not mentioning this.

I double checked it with `v3.0.0` tag and result is the same

maruel, 2018-07-16:
oops I probably messed up on the last refactoring when I increased the resolution 10x and probably missed the calculation.
https://github.com/google/periph/blob/master/devices/bmxx80/bmx280.go#L49

commit f34a1b4e (2018-07-30): bmxx80: fix humidity sensing

It regressed in 6423b54aadda465317e7dcf236fac6fb6f0e8202. :( This is due to
integer overflow. Tested against the executable compiled from periph v2.

Refactor the tests to make the error messages clearer.

Fixes #256
closed 2018-07-30 by maruel