google/periph #262: Add BigEndian CPU gohci worker

maruel, 2018-07-23:
Add one MIPS/Power gohci worker so that periph and extra do not regress on support.

maruel, 2018-08-06:
Scaling down to MIPS only.

maruel, 2018-08-07:
https://onion.io/omega2/ seems like an option but it requires external storage as git + checkout + go toolchain + build products is quite large. There's two options:
- Omega2+ plus SD card
- Omega2 plus USB stick

The challenge is likely that Omega2's 64Mb of RAM is limiting to build, Omega2+'s 128Mb would likely help performance significantly, for 2.50$ it's likely worth it.

BOM is the following, x2 for redundancy:
- 13$ Omega2+
- 15$ Expansion Dock
- 15$ Ethernet Expansion (I keep the lab wired as much as possible for reliability)
- 10$ SD card
- Shipping (free)

So 86$USD + 15% Canadian taxes.

@svenschwermer does that sounds like a good idea? Do you know alternatives?

(It bothers me that the board doesn't have a 3.3V regulator but that's for another rant)

svenschwermer, 2018-08-08:
I used a [LinkIt Smart 7688](http://wiki.seeedstudio.com/LinkIt_Smart_7688/) which uses the same SoC for my testing. There is also a [breakout board with Ethernet](http://wiki.seeedstudio.com/Breakout_for_LinkIt_Smart_7688_v2.0/). For reference, I used OpenWrt 18.06-rc1. One thing to consider with all of those boards is the ridiculously slow flash chip (at least with overlayfs). The uSD card is likely significantly faster, but I haven't checked that.

Would cross compiling be an option? That's what I did/do.

svenschwermer, 2018-08-08:
Additional notes for building the binary and the kernel config: The binaries need to be built with `GOMIPS=softfloat` while the kernel needs to be built with MIPS FPU emulation (option `CONFIG_KERNEL_MIPS_FPU_EMULATOR` in OpenWrt).

maruel, 2018-08-08:
About cross compiling:
I wrote [github.com/periph/bootstrap/cmd/push](https://github.com/periph/bootstrap/blob/master/cmd/push/main.go) to help with cross compiling, but for the CI tester, it's easier to build locally to run the on-device test, as the goal here is to do smoke testing. It could be done via a dance of "run on a linux host, rsync the binary and ssh in to run it" but I'd rather avoid this if possible. I'd want to connect a [periph-tester](https://github.com/periph/periph-tester) to it.

About the device:
Thanks for the reference! The LinkIt Smart 7688 is much more interesting to me than the Omega2. It uses the same CPU but it has a voltage regulator (wouhou!) which makes it much more plug and play. The Duo model is kinda cute too and this could probably be supported nearly first class in periph. The dock is cheaper and for real world use I wouldn't need one, which is a big plus for me. Overall I like this. From a quick check, I could get 2x 7688, 2x 7688 Duo, 2x ethernet doc shipped for 96.50$USD, which is not too bad albeit that would sent to the end of September.

About floating point:
Thanks for the warning. I knew one day my ban on floating point operations would be useful! :D That's explicitly why all types in [conn/physic](https://periph.io/x/periph/conn/physic) are based on int64. While int64 calculation on a 32 bits processor is quite slow, it's still less bad than soft float.

How much work is the setup, i.e. being able to flash a new firmware? My main concern is how much time I would have to sink in to support this. I've never built openwrt before.

svenschwermer, 2018-08-08:
It's not that bad:

1. Clone OpenWrt, checkout 18.06
2. `make menuconfig` => select your target and potentially other configs
3. `make`
4. Hookup the device (preferably via Ethernet)
5. https://openwrt.org/docs/guide-user/installation/generic.sysupgrade

maruel, 2018-08-08:
Got `bin/targets/ramips/mt76x8/openwrt-ramips-mt76x8-LinkIt7688-squashfs-sysupgrade.bin` after 65 minutes at 3.5GiB. Looks like this could be sustainable.

maruel, 2023-09-20:
While the Omega2 is still sold at https://onion.io/store/, it is not really cost competitive anymore.

I don't plan to work on MIPS support, especially given the company abandoned support for the architecture in 2021, so I will close this issue. https://en.wikipedia.org/wiki/MIPS_Technologies

If you want this to be supported, file a feature request at https://github.com/periph/host. Thanks.

closed 2023-09-20 by maruel

chmorgan, 2023-09-20:
:-(

It's how it goes for technology sometimes...

@maruel, what do you see as commercial alternatives to the Omega2?

maruel, 2023-09-20:
Good question. #periph on https://gophers.slack.com is probably a better place to discuss.