google/periph #271: sysfs-spi: graceful fallback on unidirectional I/O on buggy driver

maruel, 2018-08-08:
I want the sysfs-spi driver to not allocate memory unless explicitly needed.

Based on https://github.com/google/periph/pull/264#issuecomment-411192356, the SPI kernel driver for MediaTek MT7688AN CPU is buggy and won't work if no read buffer is provided.

For write-only SPI devices like apa102, there's no read buffer provided when doing the SPI transaction.

That causes a problem, as this means that right now trying to use APA102 on a MT7688AN probably doesn't work. A dummy read buffer should be provided to work around the buggy driver.

The optimal case would be to allocate a dummy buffer, but only when this situation is detected.

Paging @svenschwermer.

svenschwermer, 2018-08-08:
For reference, this is the driver I tested with: https://github.com/openwrt/openwrt/blob/v18.06.0-rc1/target/linux/ramips/patches-4.14/0043-spi-add-mt7621-support.patch

maruel, 2018-08-08:
Ah dang, it never occurred to me that the speed flag could only [take effect on a condition](https://github.com/openwrt/openwrt/blob/v18.06.0-rc1/target/linux/ramips/patches-4.14/0043-spi-add-mt7621-support.patch#L345), in this case if it were lower than the default speed.

[This line](https://github.com/openwrt/openwrt/blob/v18.06.0-rc1/target/linux/ramips/patches-4.14/0043-spi-add-mt7621-support.patch#L332) implies support for write-only.

maruel, 2023-09-20:
Ported to https://github.com/periph/host/issues/42.

closed 2023-09-20 by maruel