google/periph #280: BME280 code sample

hansamann, 2018-08-19:
I am just trying to use the BME280 on a rpi for the first time. 

Using the example from here in a file called temp.go
https://periph.io/device/bmxx80/

When I 'go build temp.go' it prints out this issue:
./temp.go:35: undefined: devices.Environment

I've seen some changes around devices.Environment/Environmental, but cannot figure out what exactly is the issue now. Can someone help?

maruel, 2018-08-19:
it's physic.Env, I forgot to update the example after a refactoring.
https://periph.io/x/periph/conn/physic#Env

can you test it and submit a fix to the web page via the pink link at the bottom of the page? Thanks

hansamann, 2018-08-19:
Hi @maruel, thx a lot. I just tried that and now it compiles. But when I run it (go run temp.go - temp.go is the fixed sample) I get this now:

go run temp.go
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x8046c]

goroutine 1 [running]:
panic(0x1483e0, 0x1050a010)
	/usr/lib/go-1.7/src/runtime/panic.go:500 +0x33c
periph.io/x/periph/devices/bmxx80.(*Dev).makeDev(0x10516100, 0x0, 0x0, 0x0)
	/home/pi/go/src/periph.io/x/periph/devices/bmxx80/bmxx80.go:344 +0x30
periph.io/x/periph/devices/bmxx80.NewI2C(0x76ee8000, 0x105141b0, 0x10510076, 0x0, 0x105141b0, 0x0, 0x0)
	/home/pi/go/src/periph.io/x/periph/devices/bmxx80/bmxx80.go:177 +0xe8
main.main()
	/home/pi/go/src/temp/temp.go:28 +0x1c0
exit status 2

Any idea? I've now tried the i2c.0 and i2c.1 pins and once also tried i2creg.Open("/dev/i2c-1") for pins 3 and 4, but always the same. 

Thx
Sven

hansamann, 2018-08-19:
I just ran i2cdetect -y 1:

pi@geckos:~$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- 76 -- 

I'm now using pins 3 and 4 as the other two pins did not seem to cut it. Same issue as above, though. Also ran the compiled binary with sudo, but it does not help. 

Please let me know what I can try.

hansamann, 2018-08-19:
thx so much - will try to submit the fix. In any case, the fixed version is this:

package main

import (
    "fmt"
    "log"

    "periph.io/x/periph/conn/physic"
    "periph.io/x/periph/conn/i2c/i2creg"
    "periph.io/x/periph/devices/bmxx80"
    "periph.io/x/periph/host"
)

func main() {
    // Load all the drivers:
    if _, err := host.Init(); err != nil {
        log.Fatal(err)
    }

    // Open a handle to the first available I²C bus:
    bus, err := i2creg.Open("")
    if err != nil {
        log.Fatal(err)
    }
    defer bus.Close()

    // Open a handle to a bme280/bmp280 connected on the I²C bus using default
    // settings:
    dev, err := bmxx80.NewI2C(bus, 0x76, &bmxx80.DefaultOpts)
    if err != nil {
        log.Fatal(err)
    }
    defer dev.Halt()

    // Read temperature from the sensor:
    var env physic.Env 
    if err = dev.Sense(&env); err != nil {
        log.Fatal(err)
    }
    fmt.Printf("%8s %10s %9s\n", env.Temperature, env.Pressure, env.Humidity)
}

closed 2019-01-14 by maruel