google/periph #281: doesn't support pull-up/pull-down

mCloud, 2018-08-24:
https://github.com/google/periph/blob/c091e84f02df7dc07db531cdf32af39bb1b37482/experimental/devices/mfrc522/mfrc522.go#L114

maruel, 2018-08-24:
This should work on a Raspberry Pi and Allwinner A64 and R8 processors. What hardware and OS are you using?

mCloud, 2018-08-24:
Thanks for your prompt reply. 
I use IMX6UL, linux kernel V4.1.15

maruel, 2018-08-31:
Update: I may get boards. Will update accordingly.

maruel, 2018-09-10:
I received Pico i.MX7D boards, so I can start with this and then follow up on the IMX6UL as both are using a fairly similar design.

CPUs:
- [NXP i.MX7D product page](https://www.nxp.com/products/processors-and-microcontrollers/arm-based-processors-and-mcus/i.mx-applications-processors/i.mx-7-processors/i.mx-7dual-processors-heterogeneous-processing-with-dual-arm-cortex-a7-cores-and-cortex-m4-core:i.MX7D)
  - [NXP i.MX7D reference manual](https://www.nxp.com/docs/en/reference-manual/IMX7DRM.pdf): Chapter 8: Chip IO and Pinmux; page 1599 and following.
- [NXP i.MX6UL product page](https://www.nxp.com/products/processors-and-microcontrollers/arm-based-processors-and-mcus/i.mx-applications-processors/i.mx-6-processors/i.mx-6ultralite-processor-low-power-secure-arm-cortex-a7-core:i.MX6UL)
  - [NXP i.MX6UL reference manual](https://www.nxp.com/docs/en/reference-manual/IMX6ULRM.pdf) Chapter 26 GPIO; page 1149

They both support 22K, 47K and 100K user selectable pull up and 100K pull down. That's really cool.

Boards:
- https://developer.android.com/things/hardware/imx7d
- [NXP Pico Board page](https://www.nxp.com/support/developer-resources/evaluation-and-development-boards/android-things/pico-i.mx7d-som-and-development-platform:PICO-i.MX7D-DEV-PLATFORM)

maruel, 2023-09-20:
I still have the board but given Android Things was shut down and nobody else requested support for it in the past 5 years, I'll close the request.

Please file a new issue at https://github.com/periph/host/ if you want NXP CPUs native support. Thanks.

closed 2023-09-20 by maruel