google/periph #285: Using rpi 3 as spi slave with periph

edp1096, 2018-09-08:
Hello, I'm trying to communicate with a circuit used TI msp430.

But, I found that rpi does not support spi slave mode from almost googling results & raspberry pi forum threads.

Is it possible to use rpi 3 as spi slave with periph?
I don't care whether bit-bang or not.

maruel, 2018-09-08:
Bitbang would be hard to achieve without a kernel driver, and from what I could find, there's none at the moment.

I'm not sure workaround I could suggest, as acting as a slave requires responding to commands in a timely manner, you'd probably have more chance with the PRU of a beaglebone.

closed 2018-09-08 by maruel

edp1096, 2018-09-13:
From your response, I decide to use rpi as master and works well now.
Thank you!