google/periph #288: bcm283x-gpio (GPIO17): pull cannot be used when subsystem gpiomem not initialized

hansamann, 2018-09-13:
I am running my code inside of an alpine docker container - just the compiled periph.io go binary is part of the container. The code worked similarly nicely outside the container. 

I am pretty sure I need to give access to the GPIO - but -v /sys:/sys does not seem to work. I assume as periph.io uses the faster access and not /sys?

What --device or volume do I need to add to the container for it to work?

// Lookup a pin by its number:
        p := gpioreg.ByName(PIN_HALL) 
        fmt.Printf("%s: %s\n", p, p.Function())

        // Set it as input, with an internal pull down resistor:
        if err := p.In(gpio.PullDown, gpio.BothEdges); err != nil {
            log.Fatal(err)
        }

        // Wait for edges as detected by the hardware
        go func() {
            for {
                p.WaitForEdge(-1)
               if p.Read() == gpio.High { 
                    fmt.Println("Iterations++")
                    state.Iterations++
                    promIterations.Set(state.Iterations)
               }
            }
        }()

hansamann, 2018-09-13:
Answering my own question:

--device /dev/gpiomem

maruel, 2018-09-14:
Yes. Note that this is Raspian specific. /dev/member is used on other distros.

Do you think it's worth documenting on the web site? You can use the "edit this page" link.