google/periph #303: feature: physic.Units flag.Value interface

NeuralSpaz, 2018-11-01:
Probably should have a discussion on what should be parsed.
My thoughts are the format should be compatible with the physic.units stringers.

eg: value|SI Prefix|unit(in ascii or unicode)
2.78µA or as 2.78uA as a concession for type-ability.

[see here](https://github.com/NeuralSpaz/periph/commit/babdbea0dbdca828fe05f0acd8104833a66eac1f)

NeuralSpaz, 2018-11-01:
#242 could also be done at same time.

maruel, 2018-11-01:
Yes. Points to take in account:

1. Some quantities have an obvious default unit; ElectricTension is definitely Volts, Frequency is Hz, etc.
1. Some quantities can be specified in multiple base unit; Angle (rad, deg), Temperature (K, C, F), gpio.Duty (%, fraction), etc.
1. Some units can be specified in multiple ways; '°', '°C', 'K', 'C', '°F', 'F', so the code must be tolerant to a reasonable extent, and must nudge the user towards the accepted forms in the error message.

I think we can address this challenge by first addressing 1. for the quantities that are not affected by 2. We could then create a generic function to implement this.

Then we could address 2 based on specific challenges for each quantity.

Oh and I'd prefer to not require floating point calculation, mainly due to the on-going tinygo work to MCUs without FPU.

NeuralSpaz, 2018-11-01:
I was reasoning that a little FP on start up for a flag might be ok, as the value would still be the unit's type? But I think it can be done without the FP a little more effort. To use them as a physic.unit parser I would be more cautious about the FP.

I agree and think it is important to allow all sensible errors such a "upa" and "µPa" for microPasca etc.
 
I have roughed out working flag.Value implementations and test for all the physic units, there are gaps and needs some work.

I would like to have decimals as part of the api as "-u xx.yUnits" seems more intuitive and forgiving than "-u xxy00Units"?

extract:
```golang
func (p *Power) Set(s string) error {
	r := []rune(s)
	v, n, err := parseNumber(r)
	if err != nil {
		return err
	}
	prefix := prefix(none)
	if !(n == len(r)) {
		prefix = parseSIPrefix(r[n])
	}
	scale := math.Pow10(int(prefix - nano))
	if prefix != none {
		s = string(r[n+1:])
	} else {
		s = string(r[n:])
	}
	switch s {
	case "watt", "watts", "Watt", "Watts", "W", "w":
		fallthrough
	default:
		*p = (Power)(int64(scale * v * float64(NanoWatt)))
	}
	return nil
}
func parseNumber(r []rune) (float64, int, error) {
	n := 0
	for i := 0; i < len(r); i++ {
		switch {
		case r[i] == '-':
			fallthrough
		case unicode.IsDigit(r[i]):
			fallthrough
		case r[i] == '.':
			n++
		default:
			break
		}
	}
	if n == 0 {
		return 1, 0, nil
	}
	value, err := strconv.ParseFloat(string(r[:n]), 64)
	if err != nil {
		return 0, 0, err
	}
	return value, n, err
}

type prefix int

const (
	femto prefix = -15
	pico  prefix = -12
	nano         = -9
	micro        = -6
	milli        = -3
	none         = 0
	deca         = 1
	hecto        = 2
	kilo         = 3
	mega         = 6
	giga         = 9
	tera         = 12
)

func parseSIPrefix(r rune) prefix {
	switch r {
	case 'f':
		return femto
	case 'p':
		return pico
	case 'n':
		return nano
	case 'u':
		return micro
	case 'µ':
		return micro
	case 'm':
		return milli
	case 'k':
		return kilo
	case 'M':
		return mega
	case 'G':
		return giga
	case 'T':
		return tera
	default:
		return none
	}
}
```

[rest of the code](https://github.com/NeuralSpaz/periph/tree/physicsUnits/conn/physic)

maruel, 2018-11-01:
https://golang.org/src/strconv/atof.go#L64 looks like a good algorithm to take inspiration from, especially for `parseNumber` you did above. Since all types in physic are already fixed point as int64 at most, it should be easy to adjust without FP calculation.

NeuralSpaz, 2018-11-02:
thank for the inspiration, if in doubt std lib.: P

my naive int implementation is a bit better than FP:
BenchmarkCelsiusString-4      	10000000	       140 ns/op	       5 B/op	       1 allocs/op
BenchmarkSetDistanceFloat-4   	 5000000	       332 ns/op	      16 B/op	       1 allocs/op
BenchmarkSetPowerInt-4        	 5000000	       285 ns/op	       8 B/op	       2 allocs/op

still working on nicer errors.

maruel, 2018-11-02:
Keep in mind int64->float64 is a single instruction on x64, and it has a ridiculously large number of FP units, which are ridiculously fast and leads to basically approximatively one cycle per instruction on these CPUs.
You'll get different results on armv6 vs armv8 vs x64. Here's a prior discussion on the subject; https://twitter.com/marcaruel/status/999804865432948741

IMHO these functions do not need to be micro optimized below 1µs runtime on slow CPUs.

NeuralSpaz, 2018-11-03:
^ that is crazy.
On pi zero:
```
BenchmarkCelsiusString    	 2000000	      3574 ns/op	       5 B/op	       1 allocs/op
BenchmarkSetDistanceFloat 	 2000000	      2718 ns/op	       4 B/op	       1 allocs/op
BenchmarkSetPowerInt      	 3000000	      2390 ns/op	       4 B/op	       1 allocs/op
BenchmarkDecimal          	 5000000	      1222 ns/op	       4 B/op	       1 allocs/op
BenchmarkDecimal2Int       	 3000000	      1732 ns/op	       4 B/op	       1 allocs/op
```
probably close enough imo.

NeuralSpaz, 2018-11-10:
physic.Frequency in #316
16 more units to go : )

maruel, 2018-11-10:
I think after we ironed out these edge cases:
- greek letters
- multiple units

I think the rest can be done in a large chunk

NeuralSpaz, 2018-11-11:
Sorry just notice this after #330.
I'll bulk up on all the ones that are straight forward.(same) as previous
I'll leave ones that need that need more work(conversions etc) for their own.

NeuralSpaz, 2018-12-11:
There are still some rough edges, but functionally I think #354 finishes of the flag.Value for units.

PR #354 (2018-12-12): physic: Adds flag.Value for physic.Temperature
This adds support for the flag.Value interface for Temperature unit type.
Add Set and flag doc examples.
Add Set() for Temperature.
Add Tests for Temperature Set().
Add Set() and flag.Value examples for Temperature.
Fix Fahrenheit rounding error in Example.
Fix Temperature stringer overflow for temperatures > allowed by Celsius.

This should close #303.
closed 2018-12-12 by NeuralSpaz