google/periph #326: sysfs: handle signals safely

maruel, 2018-11-10:
In issue #323 we discuss that SIGTSTP (Ctrl-Z) on linux may cause failures inside the program. I wouldn't be surprised if other signals would also interfere with the drives.

This should be fixed for all sysfs devices and smoke tested.

mame82, 2018-11-12:
As already mentioned in #323 `epoll_wait` syscall errors out for every signal.

According to the man_page of epoll_wait, `EINTR` sdhould be returned for both, a timeout and a received signal. From my observations the Go implementation of `syscall.EpollWait` returns 0 for fd count and nil for error, in case the timeout is reached, but for a signal based interrupt of `syscall.EpollWait` an error is returned (and a fd count of -1).
So it is easy to distinguish between intended timeouts and unintended interrupts caused by received signals.
Additionally using `epoll_pwait` instead of `epoll_wait` would allow filtering signals in an atomic way, according to the man page.

mame82, 2018-11-12:
Have found another solution which does the trick. Gonna file a PR once cleaned.

maruel, 2019-01-14:
This is going to be fixed as part of #380.

closed 2019-01-14 by maruel