google/periph #344: Implement driver for PC Cooler Axial Fan

kaskerd, 2018-11-18:
It would be good to provide a driver for a PC Cooler Axial Fan, or examples of how to interface with any Fan using pwm.

A few examples could be:
- how to start a fan
- how to stop a fan
- how to change the duty cycle a fan while running

NeuralSpaz, 2018-11-18:
This would depend on the type of fan , size of the fan and voltage, 3 or 4 wire  pwm or linear. In most circumstances you will need external circuitry to drive the common 12V PC fan from most embedded PC's as their IO has typically a low drive capacity and at 1.8V, 3.3V or 5V level. eg Raspberry Pi is 50mA(max) @ 3.3V.

If you just need a PWM example: it is in there, its just not showing up in godoc, which now that I know, I will fix.
```golang 
func ExamplePinOut_pwm() {
	// Make sure periph is initialized.
	if _, err := host.Init(); err != nil {
		log.Fatal(err)
	}

	// Use gpioreg GPIO pin registry to find a GPIO pin by name.
	p := gpioreg.ByName("GPIO6")
	if p == nil {
		log.Fatal("Failed to find GPIO6")
	}

	// Generate a 33% duty cycle 10KHz signal.
	if err := p.PWM(gpio.DutyMax/3, 10*physic.KiloHertz); err != nil {
		log.Fatal(err)
	}
}
```

kaskerd, 2018-11-18:
@NeuralSpaz Thanks for the example, my implementation was same as the above and glad for this confirmation.

It would be good to provide an example with a higher voltage fan,  a case would be with a Raspberry Pi using a relay. Obviously the hardware DIY part would up to the user.