google/periph #358: Any plans to implement sysfs pwm support?

chmorgan, 2018-12-20:
I may intend to submit a pull request to add this support but I'm currently using gobot for pwm support.

maruel, 2018-12-20:
Can you expand on what you mean exactly?

chmorgan, 2018-12-20:
It looks like hardware PWM support isn't provided via sysfs:

https://github.com/google/periph/blob/master/host/sysfs/gpio.go#L307

maruel, 2018-12-20:
Do you ask about using `/sys/class/pwm/` or something else?

chmorgan, 2018-12-20:
Hi @maruel, yep, isn't /sys/xxx sysfs or am I confused there?

maruel, 2018-12-20:
What kind of board are you looking for to leverage native PWM on?

chmorgan, 2018-12-20:
Hi @maruel, in this csae an Omega2+ board. 

I apologize if the support is already present, I combed the codebase, searched etc but I'm still a bit new to sysfs with PWM. I wouldn't want to waste anyones time.

maruel, 2018-12-20:
I see http://blog.oddbit.com/2017/09/26/some-notes-on-pwm-on-the-raspberry-pi/

I hadn't realize that newer kernels exposed it. That looks totally doable.

maruel, 2018-12-20:
Independent of Omega or RPi, this is doable, just need someone to add the enumeration and reconciliation with the GPIO pin numbers.

chmorgan, 2018-12-20:
@maruel yep, I was thinking it would be relatively transparent with respect to the hardware itself. I'm not at the point with periph that I could help contributing anything here yet.

maruel, 2023-09-20:
Ported to https://github.com/periph/host/issues/40.

closed 2023-09-20 by maruel