google/periph #359: Plans to implement Onewire (w1) sysfs support?

chmorgan, 2018-12-20:
It looks like sysfs supports Onewire, https://www.kernel.org/doc/Documentation/w1/w1.generic, and https://www.kernel.org/doc/Documentation/w1/w1.netlink

maruel, 2018-12-20:
Last time I looked, it looked like it required kernel mode drivers but the w1.netlink page seems to imply otherwise.

chmorgan, 2018-12-20:
Here on Omega2 I do have do to:

```
insmod w1-gpio-custom bus0=0,2,0
```

to connect it to GPIO2, but after that it shows up in the filesystem and I can do:

```
$> ls /sys/devices/w1_bus_master1
$>cat /sys/devices/w1_bus_master1/xxxxxx
# cat 28-0317007219ff/w1_slave 
9e 01 4b 46 7f ff 0c 10 8a : crc=8a YES
9e 01 4b 46 7f ff 0c 10 8a t=25875
```

mhabb, 2019-03-24:
@maruel I spent some time looking at the w1 sysfs and netlink interfaces and reading the corresponding kernel code. Unfortunately, neither interface seems to support requesting strong pull-ups after bus transactions; function [`w1_next_pullup` in drivers/w1/w1_io.c](https://github.com/torvalds/linux/blob/master/drivers/w1/w1_io.c#L455)  is not exposed to user space as far as I can tell. Because of that I don't think it is possible to implement periph's onewire API `Tx(w, r []byte, power Pullup) error` using the w1 module.

As an alternative, what do you think about adding a custom GPIO-based 1-wire master implementation to periph?

chmorgan, 2019-03-24:
@mhabb, do you mean using the gpio driver to produce 1-wire transactions? I can say that I’ve worked with an implementarion like this on esp32 and even with freertos the timing was too far out of spec to work consistently. I’d be surprised if you could do it with Linux through the gpio driver from userspace. I can provide a link to a blog post with some scope captures of what I found if you are interested. The solution ended up being the use of hardware.

Now it may be possible to work around the communication failures with retries but that’s assuming a crc is in place when writing data to devices to ensure when timing issues occur that the transition can be aborted.

maruel, 2019-03-26:
@mhabb if you are able to make something of value taking in accounts caveats noted by @chmorgan, I'm fine with a specific sysfs implementation.

mhabb, 2019-04-08:
Sorry for the delay! @chmorgan you are right, a bit banging implementation does not work well, at least not in Go. I did a few experiments but gave up because even simple commands like "read rom" failed with CRC errors at least half the time.

Instead I've been looking into using the w1 module via a netlink socket. Here's a PR: #406 @maruel I'll send a review request shortly. Please take a look and let me know what you think!

chmorgan, 2019-04-08:
I hope it wasn’t too frustrating that it didn’t work. I felt a bit
frustrated when I spent a few days figuring out the software onewire issues
with an rtos and seeing one bit of error every few thousands of bits.

I’m using gobot.io for onewire support via the w1 module in my new Linux/go
configuration btw.

mhabb, 2019-04-09:
@chmorgan could you elaborate? I can't seem to find much about onewire / w1 on gobot.io.

chmorgan, 2019-04-09:
Sorry about that, I’ve been using gobot.io for i2c and pwm control and was
confused. I used gobot.io because of a platform bug in periph that the
periph guys helpfully fixed (but after I wrapped up the code). When I get
back on that code I may switch over.

I have a simple ds18b20 implementation using sysfs directly. It’s not
really a driver as it uses the kernel ds18b20 driver. I can email it to you
if you are interested but it’s quite basic given that it opens a file and
converts the ascii values into a temperature.

PR #406 (2019-04-13): onewire: experimental netlink-based bus master for Linux
This driver adds a gpio-based 1-wire master that uses the Linux netlink
connector API to communicate with the w1 kernel module.

Major caveat: the w1 netlink API does not support setting strong
pull-ups after write operations. Devices such as the DS18B20 must be
powered externally in order to work with this driver.

Fixes #359
closed 2019-04-13 by maruel