google/periph #360: i2c: add type Addr

maruel, 2018-12-20:
The main rationale is that we could implement `i2c.Addr.Set(s string) error` method, which would implement `flag.Var` interface, which would reduce the amount of boiler plate in the executables in cmd/... In the meantime it's often specified as uint16 but it's annoying to process.

NeuralSpaz, 2018-12-20:
the code would be straight forward enough for the flag.Value 
```go
// Addr implements flag.Value
type Addr uint16

func (a *Addr) Set(s string) error {
	u, err := strconv.ParseUint(s, 0, strconv.IntSize)
	if err != nil {
		return errI2CSetError
	}
	if u > 0xffff {
		return errI2CSetError
	}
	*a = Addr(u)
	return nil
}
```
I presume we don't want to break api anywhere, so this would have to be an optional connivance, If maintainers wanted to accept and i2c.Addr into the New funcs for example.

NeuralSpaz, 2018-12-21:
closing since done by #362

closed 2018-12-21 by NeuralSpaz