google/periph #363: Raspberry Pi Zero W PWM issues

UnAfraid, 2018-12-21:
**Describe the bug**
I am trying to do PWM on LED with Raspberry PI Zero W, but no matter on which pin i try i always get
`bcm283x-gpio (GPIO19): bcm283x-dma: no channel available`

It works only on 0% and 100%

**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
	"log"
	"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/conn/gpio/gpioreg"
	"periph.io/x/periph/conn/physic"
	"periph.io/x/periph/host"
	"time"
)

func main() {
	_, err := host.Init()
	if err != nil {
		log.Fatalf("Couldn't initialize host: %s", err.Error())
	}

	greenGPIO := gpioreg.ByName("19")
	if err := greenGPIO.Out(gpio.Low); err != nil {
		log.Fatalf("Failed to parse green led to low: %s", err.Error())
	}
	go func() {
		i, err := gpio.ParseDuty("10%")
		if err != nil {
			log.Fatalf("Failed to parse duty cycle: %s", err.Error())
		}

		step := i / 2

		for {
			log.Printf("Raising PWM to: %v duty", i)
			if err := greenGPIO.PWM(i, 10*physic.KiloHertz); err != nil {
				log.Printf("Failed to set PWM: %s", err.Error())
			}

			i += step
			if i > gpio.DutyMax {
				break
			}

			time.Sleep(1000 * time.Millisecond)
		}
	}()
}
```
2. Run it.
3. See error

**Expected behavior**
I am expecting to dim the LED

**Platform (please complete the following information):**
 - OS: Raspbian Stretch Lite
 - Board Raspberry Pi Zero Wireless

**Additional context**
I am attempting to control the dimming on LEDs of this button: https://www.aliexpress.com/item/30mm-red-bi-color-RGB-led-Metal-Push-Button-Switch-momentary-latching-waterpoof-2no2nc-pushbutton-switch/32954214211.html

maruel, 2018-12-21:
Try using the pins which have native PWM; you can retrieve them with "PWM0" and "PWM1".

NeuralSpaz, 2018-12-21:
this work on my pi zero strech

```go
package main

import (
	"log"
	"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/conn/gpio/gpioreg"
	"periph.io/x/periph/conn/physic"
	"periph.io/x/periph/host"
	"time"
)

func main() {
	_, err := host.Init()
	if err != nil {
		log.Fatalf("Couldn't initialize host: %s", err.Error())
	}

	pwm0 := gpioreg.ByName("PWM1")
	if err := pwm0.Out(gpio.Low); err != nil {
		log.Fatalf("Failed to parse green led to low: %s", err.Error())
	}
	//	go func() {
	i, err := gpio.ParseDuty("10%")
	if err != nil {
		log.Fatalf("Failed to parse duty cycle: %s", err.Error())
	}

	step := i / 2

	for {
		log.Printf("Raising PWM to: %v duty", i)
		if err := pwm0.PWM(i, 10*physic.KiloHertz); err != nil {
			log.Printf("Failed to set PWM: %s", err.Error())
		}

		i += step
		if i > gpio.DutyMax {
			break
		}

		time.Sleep(1000 * time.Millisecond)
	}
	//	}()
}
```

UnAfraid, 2018-12-21:
After restart of my PI it worked for a while, but after restart of my application a couple of times got it into that state again

NeuralSpaz, 2018-12-21:
I get that too on next execution.

NeuralSpaz, 2018-12-22:
Watching the PWM out on a scope, I noticed that if there is a SIGINT (ctrl+c) that the pwm would not Halt correctly and would continue outputting the last pwm, and not release the dma? 
one possible way around this is to get the handle the signal in the program and call Halt() there as well.
eg:
```go 
...
	pwm0 := gpioreg.ByName("PWM1")
	defer pwm0.Halt()
	if err := pwm0.Out(gpio.Low); err != nil {
		log.Fatalf("Failed to parse green led to low: %s", err.Error())
	}

	var halt = make(chan os.Signal)
	signal.Notify(halt, syscall.SIGTERM)
	signal.Notify(halt, syscall.SIGINT)

	go func() {
		select {
		case <-halt:
			if err := pwm0.Halt(); err != nil {
				log.Println(err)
			}
			os.Exit(1)
		}
	}()
...
```

maruel, 2019-01-14:
Looks like user error, closing.

That said, the v4 API must make it very obvious so users hardly can make this error.

closed 2019-01-14 by maruel