google/periph #371: sysfs-gpio: use new `/sys/bus/gpiochipN`

maruel, 2019-01-07:
It's the "new" GPIO API for linux kernels above 4.8, based on ioctl on a single handle instead of having to open multiple handles. Now that devices are using later kernels (Raspbian is on 4.14), periph should enable the use of this interface when available, and fallback with the current code on older kernels.

The primary rationale is that it enables pull resistors via the constants `GPIOLINE_FLAG_OPEN_DRAIN` and `GPIOLINE_FLAG_OPEN_SOURCE`. It also enables support for reading and writing multiple lines at once. This would have many uses but the periph gpio API currently doesn't make this use-case easy to implement, so multi-pin synchronized I/O requires more thoughts.

Ref:

- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/uapi/linux/gpio.h includes the definitions
- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/tools/gpio/lsgpio.c#n102 is an example.
- libgpiod at https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/?h=v1.2 is the "official" user space library to wrap the calls, but it's not necessary.

temoto, 2019-06-24:
@maruel I've made a separate package for GPIO chardev https://github.com/temoto/gpio-cdev-go with the most open license I know, maybe you want to copy it into periph or use in another way. Thanks for periph, it's great.

maruel, 2019-06-24:
Thanks! I was trying to do an [internal refactoring](https://github.com/google/periph/compare/master...maruel:3_poll?expand=1) of the way events are managed first but I got side tracked for a few months. 😞

maruel, 2023-09-20:
Ported to https://github.com/periph/host/issues/39

closed 2023-09-20 by maruel