google/periph #405: GPIO not working on Raspberry Pi arm64

terrancewong, 2019-04-04:
**Describe the bug**
GPIO not working .

**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
        "fmt"
        "log"
        "periph.io/x/periph/conn/gpio"
        "periph.io/x/periph/conn/physic"
        //"periph.io/x/periph/conn/spi"
        //"periph.io/x/periph/conn/spi/spireg"
        "periph.io/x/periph/host"
        "periph.io/x/periph/host/rpi"
        "time"
)

func check() {
        state, err := host.Init()
        if err != nil {
                log.Fatal(err)
        }

        for _, loaded := range state.Loaded {
                fmt.Println("Driver Loaded: ", loaded.String())
        }
        for _, skipped := range state.Skipped {
                fmt.Println("Driver Skipped: ", skipped.String())
        }
        for _, failed := range state.Failed {
                fmt.Println("Driver Failed: ", failed.String())
        }
}
func testio() {
        t := time.NewTicker(100 * time.Millisecond)
        for l := gpio.Low; ; l = !l {
                fmt.Printf(".")
                rpi.P1_7.Out(l)
                rpi.P1_40.Out(l)
                <-t.C
        }
}

func main() {
        check()
        testio()
}
```

2. Run it.
3. See error

```
Driver Loaded:  sysfs-gpio
Driver Loaded:  sysfs-i2c
Driver Loaded:  sysfs-led
Driver Loaded:  sysfs-spi
Driver Loaded:  sysfs-thermal
Driver Skipped:  allwinner-gpio-pl: A64 CPU not detected
Driver Skipped:  allwinner-gpio: Allwinner CPU not detected
Driver Skipped:  bcm283x-dma: dependency not loaded: "bcm283x-gpio"
Driver Skipped:  bcm283x-gpio: bcm283x CPU not detected
Driver Skipped:  pine64: pine64 board not detected
Driver Failed:  rpi: rpi: failed to read cpu_info: strconv.ParseInt: parsing "": invalid syntax
```
and the IO does not flip

**Expected behavior**
IO should flip

**Platform (please complete the following information):**
 - OS: Ubuntu-Mate 18.04
 - Board: Raspberry Pi 3Bplus Wireless
```
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:        18.04
Codename:       bionic

Linux hostname 4.15.0-1033-raspi2 #35-Ubuntu SMP PREEMPT Fri Mar 15 15:53:58 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
```

**Additional context**

possible cause: /proc/cpuinfo has no CPU name or model

```
$ cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```

terrancewong, 2019-04-04:
More likely a bug of Ubuntu-Mate, will close this soon...

maruel, 2019-04-04:
https://github.com/google/periph/blob/master/host/rpi/rpi.go#L347 is the line that fails.

We should find another way to get the CPU revision number and select on it.

terrancewong, 2019-04-04:
also https://github.com/google/periph/blob/master/host/bcm283x/gpio.go, but this is not github.com/google/periph's fault.
I've already got my code to work, but just a temporary walk-around.

opened a ticket for Ubuntu-Mate https://bugs.launchpad.net/ubuntu-mate/+bug/1823151

maruel, 2019-04-04:
I suspect this is a patch that Raspbian made that they never upstreamed. It'd be worth looking at the delta there. As such, it's not much of a "bug" that lack of upstreaming. It'd be interesting to see what Archlinux reports.

chlunde, 2020-04-14:
@maruel this might have been fixed now by #436

maruel, 2020-04-14:
Cool thanks for checking, and thanks to @stapelberg for the actual fix.
Carl or @terrancewong , can you confirm for sure before closing this issue? 
I'll try to make another dot release in a non ridiculously long time.

maruel, 2021-05-16:
It's been a while, closing.

closed 2021-05-16 by maruel