google/periph #407: SH1106 support

pielgrzym, 2019-04-09:
**What kind of new feature are you looking for?**
(Keep the one that applies, please describe)
- Hardware: SH1106

**Do you plan to:**
- contribute an initial driver in experimental: Yes
- do all the work needed to get the driver promoted to _stable_¹: No

We know the second question looks scary. Do not worry if you reply _No_ to the
_stable_ question. It's still useful. It's still possible someone else will pick
it up and finish the work.

¹ Wonder what it takes to promote a driver as _stable_? See
https://periph.io/project/#driver-lifetime-management. A stable driver requires
the smallest API surface, good unit test code coverage, good documentation and a
page in
[https://periph.io/device/](https://github.com/periph/website/tree/master/site/content/device)

maruel, 2019-04-09:
Hi! Thanks! It looks like https://periph.io/x/periph/devices/ssd1306/ could be adapted but that would be a breaking change. There's two options:
- Make the driver generic
- Copy paste the driver

What I recommend is to start off in experimental with a copy this driver and figure out the best way from there.

pielgrzym, 2019-04-09:
Thanks for the tips - I was aiming as using ssd1306 as a template - I'm already using it in my project :) accidentally bought another LCD based on sh1106 - it should be fun making it work :)

pielgrzym, 2019-04-10:
I'm struggling to get the display to do anything :) here are my steps:

- I attached the display to raspberry pi 3 in (default) SPI mode
- I tested the display with luma python library - seems to work ok there.
- I created a dirty copy of the driver ([here is the source](https://gist.github.com/pielgrzym/07bcaee053244da841b945955827f257)). Only thing I changed is initialization of the display (line 316 of the `driver.go`).
- The code compiles fine, but I get nothing on the screen (I was hoping for corrupt screen at least ;) ). 

Any idea if I'm missing something obvious? Looking at luma library it seems ssd1306 and sh1106 only differ in initialization - or do they?

Thanks!

pielgrzym, 2019-04-10:
Ok, figured out I need to set GPIO25 to High (Reset pin) - now I get misaligned text, which is great first step :)))

pielgrzym, 2019-04-10:
Ok, I have a working driver here: https://gist.github.com/pielgrzym/da65718b5eba64483347ca35d6cc66a5
Unfortunately - if I run it standalone it seems to render the image fine. When I run it inside my project I get this spi segmentation violation:

```
2019/04/10 15:44:05 Using sh1106 driver
2019/04/10 15:44:05 Sent cmd %b[174 32 16 176 200 0 16 64 161 166 168 63 164 211 0 213 240 217 34 218 18 219 32 141 20 129 255 175]
2019/04/10 15:44:05 0 16
2019/04/10 15:44:05 1 21
2019/04/10 15:44:05 2 20
2019/04/10 15:44:05 3 26
2019/04/10 15:44:05 Sent cmd %b[176 2 16]
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x14 pc=0xf0a9c]

goroutine 1 [running]:
periph.io/x/periph/host/sysfs.(*spiConn).txPackets(0x250e6c0, 0x250e774, 0x1, 0x2, 0x0, 0x0)
	/home/pielgrzym/go/src/periph.io/x/periph/host/sysfs/spi.go:411 +0x180
periph.io/x/periph/host/sysfs.(*spiConn).Tx(0x250e6c0, 0x26988e8, 0x3, 0x3, 0x0, 0x0, 0x0, 0x0, 0x0)
	/home/pielgrzym/go/src/periph.io/x/periph/host/sysfs/spi.go:299 +0x190
overseer/sh_driver.(*Dev).sendCommand(0x250a500, 0x26988e8, 0x3, 0x3, 0xffff, 0x1)
	/home/pielgrzym/go/src/overseer/sh_driver/driver.go:414 +0x108
overseer/sh_driver.(*Dev).drawInternal(0x250a500, 0x2690000, 0x400, 0x400, 0x80, 0x40)
	/home/pielgrzym/go/src/overseer/sh_driver/driver.go:366 +0x124
overseer/sh_driver.(*Dev).Draw(0x250a500, 0x0, 0x0, 0x80, 0x40, 0x1e5048, 0x24a43a0, 0x0, 0x0, 0x0, ...)
	/home/pielgrzym/go/src/overseer/sh_driver/driver.go:192 +0x1c8
overseer/display/sh1106.(*SH1106Driver).Redraw(0x24a43c0)

```

I am completely clueless why this happens. I tried passing exactly same bytes as image from my project - I could not reproduce. Everything is the same except that in my project I'm using the driver as a different package. Any idea what could be wrong?

maruel, 2019-04-10:
The line numbers do not match https://gist.github.com/pielgrzym/da65718b5eba64483347ca35d6cc66a5.

You may want to use https://github.com/maruel/panicparse to help with stack trace readability.

Looks like something is nil.

pielgrzym, 2019-04-10:
Sorry for misaligned lines - the crash occurs on line 362 (`d.sendCommand([]byte{page_address, col & 0xf, 0x10 | (col >> 4)})`) - when sending the command via spi. The return from txPackets contains an insecure pointer - how may my code affect this pointer?

Thanks for the tool - I'll dig into this :)

pielgrzym, 2019-04-10:
Here is the ouput from panicparse:
```
[signal SIGSEGV: segmentation violation code=0x1 addr=0x14 pc=0xf0a9c]

To see all goroutines, visit https://github.com/maruel/panicparse#gotraceback

1: running
    sysfs     spi.go:411           (*spiConn).txPackets(#2, 0x152e534, 0x1, 0x2, 0, 0)
    sysfs     spi.go:299           (*spiConn).Tx(#2, #4, 0x3, 0x3, 0, 0, 0, 0, 0)
    sh_driver driver.go:414        (*Dev).sendCommand(#3, #4, 0x3, 0x3, 0xffff, 0x1)
    sh_driver driver.go:366        (*Dev).drawInternal(#3, 0x1674000, 0x400, 0x400, 0x80, 0x40)
    sh_driver driver.go:192        (*Dev).Draw(#3, 0, 0, 0x80, 0x40, 0x1e5048, 0x14c03a0, 0, 0, 0, ...)
    sh1106    sh1106.go:87         (*SH1106Driver).Redraw(#1)
    display   window_manager.go:69 (*WindowManager).CreateMenu(0x14d0280, 0x14cc3bc, 0x4, 0x1556000, 0xa, 0xd)
    display   window_manager.go:57 NewWindowManager(0x14e0300, 0x1e5790, #1, 0x15301c0, 0x152e314)
    main      overseer.go:33       main()

```

pielgrzym, 2019-04-10:
Here is the driver with lines aligned with trace below: https://gist.github.com/pielgrzym/20d1388e984e9daf243bfdb1c6cd3731

```
2019/04/10 16:01:14 Sent cmd %b[176 2 16]
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x14 pc=0xf0a9c]

To see all goroutines, visit https://github.com/maruel/panicparse#gotraceback

1: running
    sysfs     spi.go:411           (*spiConn).txPackets(#2, 0x10a8774, 0x1, 0x2, 0, 0)
    sysfs     spi.go:299           (*spiConn).Tx(#2, #4, 0x3, 0x3, 0, 0, 0, 0, 0)
    sh_driver driver.go:416        (*Dev).sendCommand(#3, #4, 0x3, 0x3, 0xffff, 0x1)
    sh_driver driver.go:368        (*Dev).drawInternal(#3, 0x127e000, 0x400, 0x400, 0x80, 0x40)
    sh_driver driver.go:193        (*Dev).Draw(#3, 0, 0, 0x80, 0x40, 0x1e5070, 0x10688e0, 0, 0, 0, ...)
    sh1106    sh1106.go:87         (*SH1106Driver).Redraw(#1)
    display   window_manager.go:69 (*WindowManager).CreateMenu(0x10a72c0, 0x107a99c, 0x4, 0x109a820, 0xa, 0xd)
    display   window_manager.go:57 NewWindowManager(0x106e100, 0x1e57b8, #1, 0x1064340, 0x10a84c8)
    main      overseer.go:33       main()

```

pielgrzym, 2019-04-10:
Figured it out :) first! sorry for bothering you with such stupid mistakes. My project closes spi in defer - unlike the file that I sent (main.go) the code is launched after deferred close, hence the spi is uninitialized and everything crashes. Now it works flawlessly :) In the next couple of days I'll give the driver some love (drawing code is taken from: https://github.com/suapapa/go_devices/tree/master/sh1106 - I need to optimize it) and I'll commit some branch with it :)

maruel, 2019-04-12:
Ok. Just be sure you actually author the code.

sztanpet, 2019-07-15:
For the record, I have also made a sh1106 driver here https://github.com/sztanpet/sh1106
the only changes that I made from the ssd1306 are:

- removed support for specifying screen sizes
- added support for specifying the i2c address
- modified drawInternal for the sh1106 memory model
- full support for the entire calculateSubset return values

This code is running perfectly in production, the display is on I2C though, and I have ever only tried it on that.
Not sure if it makes a difference if I say my code is in the public domain or not, but feel free to use any or none of it.

Thanks for the great library!

maruel, 2019-07-20:
@sztanpet Cool! I don't have strong opinion on how to proceed beside what I already stated above.

maruel, 2021-05-16:
If you still want to contribute, please send a PR against https://github.com/periph/devices.

closed 2021-05-16 by maruel