google/periph #416: [Question] How do I set a pin to alt0?

rolandjitsu, 2019-07-22:
Say I use the C lib [wiringPi](http://wiringpi.com/reference/core-functions/) and I set the following pins:
```cpp
pinMode (0, 0);
pinModeAlt (0, 0b100);
pinMode (1, 0);
pinModeAlt (1, 0b100);
pinMode (3, 1);
digitalWrite(3,1);
pinMode (2, 1);
digitalWrite(2,1);
```

I'm having trouble matching the above to this lib. I have the following so far:
```go
package hardware

import (
	"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/conn/pin"
	"periph.io/x/periph/host"
	"periph.io/x/periph/host/bcm283x"
)

func SetupI2C() error {
	// Load drivers
	if _, err := host.Init(); err != nil {
		return err
	}

	// https://pinout.xyz/pinout/pin27_gpio0
	if err := bcm283x.GPIO0.In(gpio.PullNoChange, gpio.NoEdge); err != nil {
		return err
	}

	// https://pinout.xyz/pinout/pin28_gpio1
	if err := bcm283x.GPIO1.In(gpio.PullNoChange, gpio.NoEdge); err != nil {
		return err
	}

	// https://pinout.xyz/pinout/pin3_gpio2
	if err := bcm283x.GPIO2.Out(gpio.High); err != nil {
		return err
	}

	// https://pinout.xyz/pinout/pin5_gpio3
	if err := bcm283x.GPIO3.Out(gpio.High); err != nil {
		return err
	}
}
```
But I'm not really sure how to set the alt function on the pins. I'm also not sure if this is possible to do using the lib.

maruel, 2019-07-22:
Can you explain what you are trying to achieve? We never used wiringpi.

rolandjitsu, 2019-07-24:
Ok, I'll try my best.

The whole setup is for the I2C. Basically, the first 4 lines of wiringPi code are setting the `GPIO0` and `GPIO1` to input mode and then sets the pin functionality to operational mode 4 (use as I2C).

You can check the [datasheet](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf) (page 102) for the operational modes and see this [post](https://www.raspberrypi.org/forums/viewtopic.php?t=79928) explaining what this mode `0b100` means (translates [ALT0](https://github.com/Wyliodrin/wiringPi/blob/bf84547136e977d88b69835bac69420e90d1ed8b/wiringPi/wiringPi.c#L120) or operational mode 4).

maruel, 2019-07-24:
Then what about enabling i2c1 in raspi-config then use i2creg?

maruel, 2019-07-24:
References:

- https://godoc.org/periph.io/x/periph/conn/i2c/i2creg
- https://godoc.org/periph.io/x/periph/host/rpi
- https://www.raspberrypi.org/documentation/configuration/raspi-config.md

rolandjitsu, 2019-07-25:
> Then what about enabling i2c1 in raspi-config then use i2creg?

Because I'm having a REST server that's written in Go which is supposed to have an endpoint that does the pin setup. So I wanted to use this lib to achieve that instead of calling exec.

Thanks for the references, but do you have some example illustrating how to setup some pins as I2C using i2creg?

maruel, 2019-07-25:
Doing the pin setup manually won't load the kernel driver, which is the key point of raspi-config. Anyhow here's the documentation to do what you asked for.

The function you want is https://godoc.org/periph.io/x/periph/host/bcm283x#Pin.SetFunc

Documentation is at https://godoc.org/periph.io/x/periph/conn/pin#PinFunc

Constants are at https://godoc.org/periph.io/x/periph/conn/i2c#pkg-constants

closed 2019-07-25 by maruel