google/periph #418: All pins are exported during bcm283x Init

nkovacs, 2019-08-07:
**Describe the bug**
During initialization of the bcm283x gpio driver, it exports all gpio pins using sysfs, and leaves them like that.

[`Init()` calls `Func()`](https://github.com/google/periph/blob/master/host/bcm283x/gpio.go#L1257), which [calls `Func()` in sysfs](https://github.com/google/periph/blob/master/host/bcm283x/gpio.go#L248) (`gpioMemory` will always be nil as it is initialized later in `Init()`), [which then calls `open()`](https://github.com/google/periph/blob/master/host/sysfs/gpio.go#L88), which [exports the pin](https://github.com/google/periph/blob/master/host/sysfs/gpio.go#L335).

There's no way to unexport these pins in periph.io, but that wouldn't be as big of an issue if it only exported the pins I'm actually using. In my case periph.io prevented the w1-gpio kernel module from working because my program exported gpio 4 despite only using gpio 26, so the w1-gpio module (which has to be loaded after my program) failed to load.

**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
    "periph.io/x/periph/host"
     _ "periph.io/x/periph/host/bcm283x"
)

func main() {
    host.Init()
}
```
2. Check if any gpio pin is exported, e.g. `ls -l /sys/class/gpio/gpio4` (if the w1-gpio module is already loaded, it will prevent the export of the pin it is using, which is gpio 4 by default)

**Expected behavior**
No gpio pins should be exported in `/sys/class/gpio`, or at least there should be a way to clean up.

**Platform (please complete the following information):**
 - OS: Raspbian Buster (10.0), Stretch (9.4)
 - Board Raspberry Pi 3B+

**Additional context**
Add any other context about the problem here.

maruel, 2019-08-07:
The rationale for "export all at startup" was to reduce the risk of late failure.

What you bring up is a good point. I'm fine with changing the behavior.

maruel, 2019-08-10:
Would you want to work on it?

maruel, 2021-05-16:
If one cares about this, the person should file an issue at https://github.com/periph/host.

closed 2021-05-16 by maruel