google/periph #419: Support for firmata

draeron, 2019-08-11:
I was wondering if it was possible to use periph with the [firmata](https://github.com/firmata/arduino) protocol? It would be very convenient if you want to use golang with board such as the ESP32 or ESP8266 which aren't really supported. This could unlock a lot of possible devices.

maruel, 2019-08-12:
Official documentation: https://github.com/firmata/protocol

Gobot has a minimal implementation (without SPI): https://godoc.org/gobot.io/x/gobot/platforms/firmata

There's another 5 years old precedent: https://godoc.org/github.com/kraman/go-firmata which includes SPI. It is not usable as-is but it gives an idea of how much code this represents.

It's totally doable, just needs someone to create the implementation. The current lack of formal UART support may be limiting.

maruel, 2021-05-16:
If someone wants to implement this, please file a PR against https://github.com/periph/devices.

closed 2021-05-16 by maruel