google/periph #420: gpio-read: bcm283x-gpio (GPIO7): pull cannot be used when subsystem gpiomem not initialized.

crossan007, 2019-08-14:
**Describe the bug**
A clear and concise description of what the bug is.
While attempting to read from a GPIO pin on a RaspberryPi 1, I receive an error: `gpio-read: bcm283x-gpio (GPIO7): pull cannot be used when subsystem gpiomem not initialized.`

**To Reproduce**
Steps to reproduce the behavior:
1. Run CLI commands to use periph tools
```
go get periph.io/x/periph/cmd/...
cd ~/go/bin
./gpio-read GPIO7
```
2. Run it.
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect the pin to be set to read-mode, and a value returhed

**Platform (please complete the following information):**
 - OS: Raspbian GNU/Linux 10 (buster)
 - Board: RaspberryPi 1

This looks similar to #288, but I'm not running my Go code inside a docker container - just the plain logged-in user.

I've also tried "granting" access to the devices listed in #288 via:
```
sudo chmod a+rw /dev/gpiomem
sudo chmod a+rw /dev/mem
```
and I still get the error.

crossan007, 2019-08-14:
I get the same error if I try to run a simple button press detection code:
```
package main

import (
    "fmt"
    "log"

    // "periph.io/x/periph/host/rpi"
    "periph.io/x/periph/conn/gpio"
    "periph.io/x/periph/conn/gpio/gpioreg"
    "periph.io/x/periph/host"
)

func main() {
    // Load all the drivers:
    if _, err := host.Init(); err != nil {
        log.Fatal(err)
    }

    // Lookup a pin by its number:
    //p := rpi.P1_26
    p := gpioreg.ByName("GPIO7")
    if p == nil {
        log.Fatal("Failed to find pin 26")
    }

    fmt.Printf("%s: %s\n", p, p.Function())

    // Set it as input, with an internal pull down resistor:
    if err := p.In(gpio.PullDown, gpio.BothEdges); err != nil {
       log.Fatal(err)
    }

    // Wait for edges as detected by the hardware, and print the value read:
    for {
        p.WaitForEdge(-1)
        fmt.Printf("-> %s\n", p.Read())
    }
}
```

NeuralSpaz, 2019-08-14:
Have you added the host gpiomem device to the container?
ie: `docker run --device /dev/gpiomem -d foo`

crossan007, 2019-08-14:
@NeuralSpaz I'm *not* running in a docker container here.

crossan007, 2019-08-14:
additionally `./periph-info` shows:
```
Drivers failed to load and the error:
- bcm283x-gpio: gpioreg: can't register an alias with no name
``

maruel, 2019-08-14:
Seems like a regression on RPi1. I'll find mine and will try to reproduce tomorrow morning.

My hunch is likely this line; https://github.com/google/periph/blob/master/host/bcm283x/gpio.go#L1265

crossan007, 2019-08-14:
@maruel Thanks for the quick reply!  Let me know if there's any information I can provide to help.

I'll note that after a "fresh boot", my program takes ~4 seconds to fail *the first time*, but on subsequent executions, fails almost instantly.

NeuralSpaz, 2019-08-14:
I was planing to do a rpi1 with buster later today, I see what I can reproduce.

maruel, 2019-08-14:
Can you try to make RegisterAlias() panic instead?

At https://github.com/google/periph/blob/master/conn/gpio/gpioreg/gpioreg.go#L108

Add:
`panic(dest)`

Then paste the stack trace here.

crossan007, 2019-08-14:
```
pi@raspberrypi:~/go/src/button $ ./button
panic: GPIO16

goroutine 17 [running]:
periph.io/x/periph/conn/gpio/gpioreg.RegisterAlias(0x0, 0x0, 0x141323, 0x6, 0x0, 0x0)
        /home/pi/go/src/periph.io/x/periph/conn/gpio/gpioreg/gpioreg.go:108 +0x3b8
periph.io/x/periph/host/bcm283x.(*driverGPIO).Init(0x223488, 0x14bac8, 0x2412e60, 0x0)
        /home/pi/go/src/periph.io/x/periph/host/bcm283x/gpio.go:1265 +0x36c
periph.io/x/periph.(*stage).loadParallel.func1.1(0x2412e60, 0x24180c0, 0x24181c0, 0x2418140, 0x2418100, 0x14368c, 0xc, 0x166a28, 0x223488)
        /home/pi/go/src/periph.io/x/periph/periph_parallel.go:84 +0x48
created by periph.io/x/periph.(*stage).loadParallel.func1
        /home/pi/go/src/periph.io/x/periph/periph_parallel.go:82 +0x304
```

maruel, 2019-08-14:
Ok so the pin is set to a function between 0 and 2. The ones that are not defined at:
https://github.com/google/periph/blob/master/host/bcm283x/gpio.go#L939

maruel, 2019-08-14:
https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf page 102:

- 0 is reserved
- 1 is SD8
- 2 is reserved

But the code in `Func()` shouldn't return an empty string
It feels like this line triggers but I don't know how;
https://github.com/google/periph/blob/master/host/bcm283x/gpio.go#L292

Can you add `panic(f)` ?

crossan007, 2019-08-14:
Doesn't look like that line(292) gets hit.  

The previous panic was hit first (https://github.com/google/periph/blob/master/conn/gpio/gpioreg/gpioreg.go#L108), so I cleared it, and now I'm back at the original error.

maruel, 2019-08-14:
Ok thanks for the debugging, I need to log off now will try to take a look tomorrow to find the root cause.

crossan007, 2019-08-14:
I amended https://github.com/google/periph/blob/master/host/bcm283x/gpio.go#L1263 as follows (beginning on line 1263) to include logging as it creates the aliases:
```
if _, ok := functions[f]; !ok {
    functions[f] = struct{}{}
    log.Println("Registering Alias: ",string(f),name)
    if err := gpioreg.RegisterAlias(string(f), name); err != nil {
            log.Println("Error registering alias: ", err)
            return true, err
    }
}
```

The output was somewhat unexpected:
```
pi@raspberrypi:~/go/src/button $ ./button
2019/08/14 02:29:47 Registering Alias:  In/High GPIO0
2019/08/14 02:29:47 Registering Alias:  Out/Low GPIO5
2019/08/14 02:29:47 Registering Alias:  Out/High GPIO6
2019/08/14 02:29:47 Registering Alias:  In/Low GPIO9
2019/08/14 02:29:47 Registering Alias:   GPIO16
2019/08/14 02:29:47 Error registering alias:  gpioreg: can't register an alias with no name
GPIO7: In/High
2019/08/14 02:29:47 bcm283x-gpio (GPIO7): pull cannot be used when subsystem gpiomem not initialized
```

are the aliases created in a random order?

The output _does_ confirm your statement about https://github.com/google/periph/blob/master/host/bcm283x/gpio.go#L939 being the culprit.

I too am signing off for the evening, Cheers!

smalld64, 2019-08-14:
Same problem I believe. First attempt at using Periph. Raspberry Pi Zero simple button input example.
GPIO22: In/High
2019/08/14 16:25:08 bcm283x-gpio (GPIO22): pull cannot be used when subsystem gpiomem not initialized

maruel, 2019-08-14:
Sorry for the annoyance. I broke this in v3.2.0. I fixed it, added a smoke test and manually ran the smoketest on a RPi1 and RPiZW. This should be fixed for good. Will try to do a v3.6.1 release next week or so.

closed 2019-08-14 by maruel

crossan007, 2019-08-14:
Thanks for taking care of it

Since I'm a bit of a Go newb, will re-running `go get ...` get the latest `master` version, or the latest release? 

can I get ebc08ac via go get?

maruel, 2019-08-14:
It depends which Go version you are using and if you enabled [GO111MODULE](https://blog.golang.org/modules2019). If not, a `go get -u periph.io/x/periph` should do the trick.

NeuralSpaz, 2019-08-15:
confirmed working on rpi1B running buster as non root.

smalld64, 2019-08-15:
Thanks working on raspberry pi zero as non root.

crossan007, 2019-08-16:
🎉 It works for me as well.  Thanks!

commit ebc08ac5 (2019-08-14): bcm283x: fix invalid aliasing; causing failure on RPi1 and RPi Zero

The regression was introduced in
3d9b008db17eda515b81dff5c4cf9c918339fbaa, so version v3.2.0 up to v3.6.0
are affected.

Add smoketest to catch regressions. Manually tested on a RPi1 and RPiZW.

Fixes #420