google/periph #422: Support for BME680

mstahl, 2019-10-14:
**What kind of new feature are you looking for?**
(Keep the one that applies, please describe)
- Hardware: BME680 temperature/humidity/pressure/VOC sensor

The `bmxx80` driver doesn't support the BME680, which is just like the 280 that is supported, but with a VOC sensor added to it. There's not much more to it in addition to the features that are already supported; I think I'd just need to add support for the VOC sensor and for the library to recognize a BME680.

**Do you plan to:**
- contribute an initial driver in experimental: Yes
- do all the work needed to get the driver promoted to _stable_¹: Yes

We know the second question looks scary. Do not worry if you reply _No_ to the
_stable_ question. It's still useful. It's still possible someone else will pick
it up and finish the work.

¹ Wonder what it takes to promote a driver as _stable_? See
https://periph.io/project/#driver-lifetime-management. A stable driver requires
the smallest API surface, good unit test code coverage, good documentation and a
page in
[https://periph.io/device/](https://github.com/periph/website/tree/master/site/content/device)

maruel, 2019-10-17:
Thanks! Please modify bmxx80 directly. The Sense() function is not going to work to pass VOC so add another separate method for now. I happen to have bought some, so I'll be able to test immediately, I just didn't have time to do the code.

mstahl, 2019-10-17:
Great! I started by modifying the `bmxx80` driver directly anyway, figuring I could extract that later if needed. Have been focusing on just getting accurate readings for the already-supported values (temp/humidity/pressure). Doing VOC as a separate function makes sense as long as the chip doesn't want to do all of them in one shot like the other measurements. I'll have some time next week to work on this.

maruel, 2021-05-16:
I am still blocked on refactoring that I wanted to do in v4 but had trouble finding time to complete. This issue would be reopened at https://github.com/periph/devices.

closed 2021-05-16 by maruel