google/periph #423: Halt() does not unblock WaitForEdge()

toxygene, 2019-10-14:
According to the documentation, calling `Halt()` or `In()` on a pin should cause `WaitForEdge()` to unblock and return false. When I attempt to do this on a Raspberry Pi Zero W, however, `WaitForEdge()` does not become unblocked.

```go
package main

import (
	"fmt"
	"time"

	"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/conn/gpio/gpioreg"
	"periph.io/x/periph/host"
)

func main() {
	if _, err := host.Init(); err != nil {
		panic(err)
	}

	aPin := gpioreg.ByName("23")
	aPin.In(gpio.PullUp, gpio.BothEdges)

	go func() {
		time.Sleep(1 * time.Second)

		fmt.Println("calling Halt()")
		aPin.Halt()
		fmt.Println("called Halt()")
	}()

	fmt.Println("wait for edge")
	aPin.WaitForEdge(-1)
	fmt.Println("done waiting for edge")
}
```

This program will output:
```
wait for edge
calling Halt()
called Halt()
```
and then block indefinitely.

**Platform (please complete the following information):**
 - OS: Raspbian GNU/Linux 9
 - Board: Raspberry Pi Zero Wireless

maruel, 2019-10-17:
Sorry I didn't have time to play with this locally and won't likely have time until this weekend. :/

toxygene, 2019-10-17:
> Sorry I didn't have time to play with this locally and won't likely have time until this weekend. :/

Don't worry about it. I don't need it for anything critical -- it's only for a small side project I'm working on.

maruel, 2023-09-18:
Manually ported to https://github.com/periph/host/issues/36.

closed 2023-09-18 by maruel