google/periph #424: Feature SSD1322

MarkoBursic, 2019-11-05:
Is there a plan to support also the display SSD1322 via SPI 3/4-wire, soon?

maruel, 2019-11-05:
To be clear, I appreciate contributions to extend hardware support. :)

MarkoBursic, 2019-11-05:
I understand. Can you give a brief info on how you proceeded writing SSD1306, perhaps an example written in different language that you ported in your project. SSD1322 seems to be very close, so many of your code might be reused.

maruel, 2019-11-08:
I always start from the datasheet, never from another drivers. There's multiple reasons for this.

If you can make the other driver more generic go for it. I seem to recall someone implementing a 4 bits image and driver but I forget if it was merged in or living out of tree; I'm currently in transit so can't confirm.

maruel, 2021-05-16:
If you want to work on this, please contribute to https://github.com/periph/devices.

closed 2021-05-16 by maruel