google/periph #426: spi to slow

lclslsls, 2019-11-26:
**Describe the bug**
i use the spi to read the adc 
but the spi is to slow 

[image]

i need at lest 100k hz 
but now it just 10k hz
**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go

import (
	"fmt"
	"log"
	"io"
	//"time"
//	"encoding/binary"
	//"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/conn/physic"
	"periph.io/x/periph/conn/spi"
	"periph.io/x/periph/conn/spi/spireg"
	"periph.io/x/periph/host"
)

func main() {
	//f := ioctlClose{}
	// Make sure periph is initialized.
	if _, err := host.Init(); err != nil {
		log.Fatal(err)
	}
	// Using SPI as an example. See package ./spi/spireg for more details.
	p, err := spireg.Open("/dev/spidev32766.0")
	if err != nil {
		log.Fatal(err)
	}
	defer p.Close()
	c, err := p.Connect(5*physic.MegaHertz, spi.Mode0, 32)
	if err != nil {
		log.Fatal(err)
	}
//0x0d 代表写入命令
//0x14 代表 写入输出控制寄存器
//0x02 代表 正负5v 输出{0x02,0x00,0x0x14,0xd0}
	commod_Set_Fulrange := []byte{0x03,0x00,0x14,0xd0}//将采样范围设置到 正负5v
	log.Println("len:",len(commod_Set_Fulrange))
	read := make([]byte, len(commod_Set_Fulrange))
	if err := c.Tx(commod_Set_Fulrange, read); err != nil {
		log.Fatal(err)
	}
	write := []byte{0x00,0x00,0x00,0x00}
	read2 := make([]byte, len(write))
	for i:=0;i<1024;i++{
		c.Tx(write, read2)
	}
	

}
2. Run it.
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
it works but to slow
**Platform (please complete the following information):**
 - OS: [e.g. Raspbian Stretch Lite]
 - Board [e.g. Raspberry Pi Zero Wireless]

**Additional context**
Add any other context about the problem here.

maruel, 2019-11-26:
Sadly the SPI driver will never give you an exact match of the clock frequency requested, always something slower based on the current clock. I haven't found a way to fix this.

I recommend iterating over the specified frequency going up until you get in practice one that you like.

lclslsls, 2019-11-27:
> Sadly the SPI driver will never give you an exact match of the clock frequency requested, always something slower based on the current clock. I haven't found a way to fix this.
> 
> I recommend iterating over the specified frequency going up until you get in practice one that you like.

the clock frequency  is ok,
my quez is that when one frame is over the ce cost too much time to transfer the next frame
just like the picture show

maruel, 2019-11-27:
Ah I think I understand now, you mean that the delay between requests is too large? AFAIK all of this time is spent in the kernel. Tx() leads to exactly a single system call:

https://github.com/google/periph/blob/8dc522fd6d6756e1ee5f72c23f05a06a6382776c/host/sysfs/spi.go#L411

lclslsls, 2019-11-27:
> Ah I think I understand now, you mean that the delay between requests is too large? AFAIK all of this time is spent in the kernel. Tx() leads to exactly a single system call:
> 
> https://github.com/google/periph/blob/8dc522fd6d6756e1ee5f72c23f05a06a6382776c/host/sysfs/spi.go#L411
you are right ,i write a c demo 
it has same quiz  
thank you very much 
by the way is there some way to make ioct more faster??

closed 2019-11-27 by lclslsls

maruel, 2019-11-27:
Not as far as I know. I suspect it's highly driver specific. One thing I never tried is to use NoCS, and manage it yourself. Maybe you'd save a bit on the timeouts.