google/periph #427: onewire didnot work

lclslsls, 2019-12-12:
**Describe the bug**
A clear and concise description of what the bug is.
./onewire 
onewirereg2019/11/05 14:08:54 onewirereg: no bus found; did you forget to call Init()?
**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
  "periph.io/x/periph/host"
)

func main() {
if _, err := host.Init(); err != nil {
		
		log.Fatal(err)
	}

	// Use onewirereg 1-wire bus registry to find the first available 1-wire bus.
	b, err := onewirereg.Open("")
	if err != nil {
		fmt.Printf("onewirereg")
		log.Fatal(err)
	}
	defer b.Close()

	// Prints out the gpio pin used.
	if p, ok := b.(onewire.Pins); ok {
		fmt.Printf("Q: %s", p.Q())
	}
}
```
2. Run it.
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
 ls /sys/bus/w1/devices/                  
28-0000079e36c0  w1_bus_master1
the bus is fine and it can read by shell
cat /sys/bus/w1/devices/28-0000079e36c0/w1_slave 
f4 01 4b 46 7f ff 0c 10 c7 : crc=c7 YES
f4 01 4b 46 7f ff 0c 10 c7 t=31250
just the golang demo code cannot run 
**Platform (please complete the following information):**
 - OS: [e.g. Raspbian Stretch Lite]
NXP i.MX Release Distro 4.9.88-2.0.0
 - Board [e.g. Raspberry Pi Zero Wireless]

**Additional context**
Add any other context about the problem here.

maruel, 2019-12-12:
See https://periph.io/x/periph/experimental/host/netlink

lclslsls, 2019-12-13:
I swa the link 
it seems i used the wrong method 
but by this link i cannt find something like testcode or democode 
i didnot now how to use it  
is some where has democode?

maruel, 2019-12-13:
You'll have to ask the code authors, as I personally didn't work on this code. https://github.com/google/periph/pull/406

lclslsls, 2019-12-16:
but why the onewire test code didnot work 
how can i fix this 
the spi is work perfect

maruel, 2019-12-16:
I can't parse your question.

I said exactly what to do, you'll have to ask the code authors. I said explicitly I did not work on this code, I never used it.

rolandjitsu, 2019-12-17:
I've actually seen the same error when using the serial package. Couldn't figure out why though.

bh90210, 2020-02-20:
hi all,

I had the same issue and since I figured it out I thought of sharing an example here.

```go
package main

import (
	"fmt"
	"os"
	"os/signal"
	"syscall"
	"time"

	"periph.io/x/periph/devices/ds18b20"
	"periph.io/x/periph/experimental/host/netlink"
	"periph.io/x/periph/host"
)

func main() {
	host.Init()

	// get 1wire bus
	oneBus, _ := netlink.New(001)

	// get 1wire address
	addr, _ := oneBus.Search(false)

	// init ds18b20
	sensor, _ := ds18b20.New(oneBus, addr[0], 10)

	ds18b20.ConvertAll(oneBus, 10)
	temp, _ := sensor.LastTemp()

	fmt.Println(temp)
}
```

maruel, 2021-05-16:
The code was migrated to periph.io/x/host/v3/netlink. If you want to contribute please file a PR or an issue on https://github.com/periph/host.

closed 2021-05-16 by maruel

cgossain, 2022-02-13:
Hi all,

I had the same issue. The answer posted by [bh90210](https://github.com/google/periph/issues/427#issuecomment-589421307) above really helped me out! 

However, I just wanted to post an updated implementation that references the packages at the new locations.

The following example simply prints the discovered 1-wire devices.

```
package main

import (
	"fmt"
	"log"

	"periph.io/x/host/v3"
	"periph.io/x/host/v3/netlink"
)

func main() {
	// make sure periph is initialized
	_, err := host.Init()
	if err != nil {
		log.Fatal(err)
	}

	// get the 1-wire bus (via Netlink)
	w1, err := netlink.New(001)
	if err != nil {
		log.Fatal(err)
	}

	// get 1-wire addresses
	addrs, err := w1.Search(false)
	if err != nil {
		log.Fatal(err)
	}

	// print the 1-wire device addresses
	for i, adr := range addrs {
		fmt.Println(i, adr)
	}
}
```