google/periph #429: Experimental serial package doesn't export essential func

rolandjitsu, 2019-12-18:
**Describe the bug**
It seems like [serial.go](https://github.com/google/periph/blob/master/experimental/host/serial/serial.go#L52) doesn't export the one function needed to setup a serial conn.

After a brief look at the file, it seems like `newPortDevFs` should probably be `NewPortDevFs` so that we can use it to create a serial conn.

At the moment, there is no way to actually do it. The only thing you can do is create the `Port` struct, but that is obviously pretty useless as illustrated below.

**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
  "log"

  "periph.io/x/periph/host"
  "periph.io/x/periph/conn/physic"
  "periph.io/x/periph/experimental/conn/uart"
  "periph.io/x/periph/experimental/host/serial"
  "periph.io/x/periph/host"
  "periph.io/x/periph/host/rpi"
)

func main() {
  host.Init()
  p := serial.Port{} // Here I would use the NewPortDevFs() func 
  if _, err := p.Connect(19200*physic.Hertz, uart.One, uart.NoParity, uart.NoFlow, 8); err != nil {
    log.Fatalln(err)
  }
}
```
2. Run it.
3. See error
```
2019/12/18 01:51:24 sysfs-uart: already closed
```

**Expected behavior**
I should be able to create a serial conn without errors.

I could make a PR, but I'm not sure if that func is intentionally left out.

**Platform (please complete the following information):**
 - OS:
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
 - Board: [CM3+](https://www.raspberrypi.org/products/compute-module-3-plus)

**Additional context**
Add any other context about the problem here.

rolandjitsu, 2019-12-18:
On a side note, it would be nice to allow connecting on ports other than `ttyS%d` (e.g `ttyUSB0`).

maruel, 2019-12-18:
I didn't have time to work on this for a long time, would you be willing to contribute?

rolandjitsu, 2019-12-20:
@maruel I could. I don't have much time myself, but I'll try to make some.

maruel, 2021-05-16:
The API has been moved to periph.io/x/conn/v3.

closed 2021-05-16 by maruel

rolandjitsu, 2021-05-19:
> The API has been moved to periph.io/x/conn/v3.

Has there been changes to the interface? Does this experimental pkg work now?

maruel, 2021-05-19:
See https://periph.io/news/2020/a_new_start/

rolandjitsu, 2021-05-25:
> See https://periph.io/news/2020/a_new_start/

I see. Thanks for sharing this!