google/periph #432: Unable to set GPIO pin pull on Raspberry Pi 4

alethenorio, 2020-01-15:
**Describe the bug**
I am trying to configure a GPIO pin on the raspberry pi 4, more specifically the Pin number 29 (Connected to GPIO 5), and set it to pull down however when reading the pin it reports pull up.

This works fine on a raspberry pi 3

**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
	"fmt"
	"log"
	"time"

	"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/host"
	"periph.io/x/periph/host/rpi"
)

var (
	readPin = rpi.P1_29
)

func main() {
	fmt.Println("Initializing periph host...")
	if _, err := host.Init(); err != nil {
		log.Panicf("failed to initialize periph host: %v", err)
	}

	if err := readPin.In(gpio.PullDown, gpio.NoEdge); err != nil {
		log.Panicf("failed to set read pin to In(): %v", err)
	}

	ticker := time.NewTicker(2 * time.Second)
	want := gpio.Low

	for {
		select {
		case <-ticker.C:
			time.Sleep(time.Millisecond)
			actual := readPin.Read()
			if actual != want {
				log.Printf("ERROR: Actual: %t != Wanted: %t", actual, want)
				continue
			}
			log.Println("EVERYTHING WAS FINE")
		}
	}
}
```
2. Run it.
3. See error
`Actual: true != Wanted: false`

**Expected behavior**
It should set the Pin 29 to Pull Down (Meaning I get `gpio.Low` when reading it) thus printing only `EVERYTHING WAS FINE`

**Platform (please complete the following information):**
 - OS: balenaOS 2.44.0+rev3 (64-bit BETA)
 - Raspberry Pi 4

**Additional context**
Compilation command: `GOOS=linux GOARCH=arm64 CGO_ENABLED=0 go build main.go`

The code has been tested on several raspberry pi 3 without any issues

maruel, 2020-01-15:
I've heard multiple reports of issues on RPi4. I'll try to find time to investigate myself but any help on this side (including a PR) is appreciated.

maruel, 2020-02-25:
I still haven't gotten time to setup my local installation. I haven't forgotten, just a bit time strapped at the moment. Sorry for the delay.

alethenorio, 2020-02-25:
No worries @maruel any help is appreciated. Thank you for taking the time to comment and update the matter

stapelberg, 2020-04-08:
Also running into this (I think) using 2020-02-13-raspbian-buster-lite.img in 64-bit mode (i.e. `arm_64bit=1` in `/boot/config.txt`). 

I have tracked it down to `/dev/gpiomem` not being present making the difference: after `sudo mv /dev/gpiomem /dev/O.gpiomem`, I can no longer run my program as unprivileged user. When using sudo, the program starts, but prints incorrect values: 
```
default pull = PullUp, pull = PullNoChangeLow
Low
```

instead of the expected:
```
default pull = PullUp, pull = PullNoChangeHigh
High
```

On the Raspberry Pi 3, the same image+program works fine. So, the issue seems to be specific to using /dev/mem (not /dev/gpiomem) on the Raspberry Pi 4.

Hope that helps in tracking it down

bezineb5, 2020-05-29:
I just had a look at pi-blaster which got updated for RPi4, and indeed there are differences in base addresses and DMA channels:
https://github.com/sarfata/pi-blaster/pull/110/files

maruel, 2020-05-29:
Thanks for the reference. Anyone knows of an updated datasheet?

maruel, 2020-05-29:
Yeah I should have "tried to search on my favorite search engine first":

- https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/bcm2711
- https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf

It's cool it's everything that is needed.

berryboy2012, 2020-05-31:
It seems that BCM2711 uses a completely different pullup/pulldown method. I think we can use RPi.GPIO as a reference https://sourceforge.net/p/raspberry-gpio-python/tickets/168/

stapelberg, 2020-06-01:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=b1d84a3d0a26c5844a22bc09a42704b9371208bb was included in Linux 5.7 (just released). Not sure if that is relevant here, or if it just makes a few more GPIO pins accessible, independent of basic GPIO functionality.

maruel, 2020-06-03:
Thanks for the reference @stapelberg . I don't think the higher ones would be usable though, as the ones over 47 for the SD card, thus accessing them is fairly dangerous. That's why bcm283x only exposes pins up to 46. Maybe the 54-57 could be safe, I dunno.

closed 2020-06-12 by maruel

stapelberg, 2020-06-12:
@maruel Could you cut a new release that includes this commit, too, please?

I’d like to publish an article about using GPIOs with periph.io on top of gokrazy soon :)

maruel, 2020-06-12:
I asked on the slack channel if people can report back if nothing is broken, as the CPU detection changed significantly. I'll cut a new tag soon.

maruel, 2020-06-15:
I didn't get feedback but I still tagged v3.6.4 with the commit. I didn't have time to test personally but I don't want to make you wait more.

stapelberg, 2020-06-15:
Thanks very much! Can confirm that it works. Published https://gokrazy.org/userguide/gpio/ :)

alethenorio, 2020-06-17:
Thank you for the fix. You guys rock. :muscle:

PR #445 (2020-06-12): bcm283x:Add support for internal pull resistor used in BCM2711
Fixes  #432