google/periph #458: I2C documentation not clear on read/write operation

soypat, 2021-02-08:
**Describe the bug**
I2C documentation lacking on how Read vs Write bit is set.

**To Reproduce**
Start writing a program that reads a Dev register. You may come across various interpretations of current functions.

*The comments on the functions below show opinions of author. No code has been tested*.

```go
// Which one returns the device registry value?

// this would seem to cause a write operation to kick in before the read operation
func ReadByte1(d *i2c.Dev, addr byte) byte {
	txb := make([]byte, 1)
	d.Tx(txb, []byte{addr}) 
	return txb[0]
}

// this seems likely to work but is confusing since a len(writebuffer)!=0 indicates a write operation
func ReadByte2(d *i2c.Dev, addr byte) byte {
	rxb := make([]byte, 1)
	d.Tx([]byte{addr},rxb) 
	return rxb[0]
}

// I believe this panics. See lines 116 and 125 from periph/host/sysfs/i2c.go
func ReadByte3(d *i2c.Dev, addr byte) byte {
	rxb := make([]byte, 2)
	rxb[0] = addr
	d.Tx([]byte{},rxb) 
	return rxb[1]
}
```

I can't find anything about how the Read/Write bit is set in the Address frame.

maruel, 2021-05-16:
Hi!

I will need more information to help you. I suspect you are working with a I²C device that exposes 8 bit registers. If so, you may want to use https://periph.io/x/conn/v3/mmr which is designed expressly for this

soypat, 2021-05-16:
Yes, it has been solved. Thanks

closed 2021-05-16 by soypat