google/periph #459: MFRC522 - Problem reading Tag

UdoWeberJR, 2021-04-07:
**Describe the bug**
I have two different typ of tags, a [chip](https://www.amazon.de/gp/product/B074S8MRQ7/) card and [stickers](https://www.amazon.de/gp/product/B081GW4X3W). When I use the chip card, the behavior is as expected. When I try to use the stickers I get the error "mfrc522: wrong number of bits 4" with changing number of bits between 4 and 7.
I build a small example to show my settings. I tried "commands.PICC_AUTHENT1B" as well as "commands.PICC_AUTHENT1A" for authentication mode.

With my smartphone I'm able to read both tags, with the [MIFARE Classic Tool from IKARUS Projects](https://play.google.com/store/apps/details?id=de.syss.MifareClassicTool&hl=de&gl=US). So the default key (FFFFFFFFFFFF) seem to be correct. I'm also able to write data to both tags with this app.

I already tried a lower and a higher antenna gain without success. 
I also tried to use a different reader on another raspberry with the same result. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
	"fmt"
	"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/conn/gpio/gpioreg"
	"periph.io/x/periph/conn/spi"
	"periph.io/x/periph/conn/spi/spireg"
	"periph.io/x/periph/experimental/devices/mfrc522"
	"periph.io/x/periph/experimental/devices/mfrc522/commands"
	"periph.io/x/periph/host"
	"time"
)

// mfrc522 rfid device
var rfid *mfrc522.Dev

// spi port
var port spi.PortCloser

// pins used for rest and irq
const (
	resetPin = "P1_22" // GPIO 25
	irqPin   = "P1_18" // GPIO 24
)

/*
Setup inits and starts hardware.
*/
func setup() {
	var err error

	// guarantees all drivers are loaded.
	if _, err = host.Init(); err != nil {
		fmt.Println(err)
	}

	// get the first available spi port (empty string).
	port, err = spireg.Open("")
	if err != nil {
		fmt.Println(err)
	}

	// get GPIO rest pin from its name
	var gpioResetPin gpio.PinOut = gpioreg.ByName(resetPin)
	if gpioResetPin == nil {
		fmt.Println("Failed to find %v", resetPin)
	}

	// get GPIO irq pin from its name
	var gpioIRQPin gpio.PinIn = gpioreg.ByName(irqPin)
	if gpioIRQPin == nil {
		fmt.Println("Failed to find %v", irqPin)
	}

	rfid, err = mfrc522.NewSPI(port, gpioResetPin, gpioIRQPin, mfrc522.WithSync())
	if err != nil {
		fmt.Println(err)
	}

	// setting the antenna signal strength, signal strength from 0 to 7
	rfid.SetAntennaGain(5)

	fmt.Println("Started rfid reader.")
}

func stringIntoByte16(str string) [16]byte {
	var data [16]byte
	copy(data[:], str) // copy already checks length of str
	return data
}

func main() {
	setup()

	// trying to read UID
	data, err := rfid.ReadUID(5 * time.Second)
	if err != nil {
		fmt.Println(err)
	} else {
		fmt.Println(data)
	}

	// trying to write data
	err = rfid.WriteCard(5*time.Second, byte(commands.PICC_AUTHENT1B), 2, 0, stringIntoByte16("Hallo Welt"), mfrc522.DefaultKey)
	if err != nil {
		fmt.Println(err)
	} else {
		fmt.Println("Write OK")
	}

	// trying to read data
	data, err = rfid.ReadCard(5*time.Second, commands.PICC_AUTHENT1A, 2, 0, mfrc522.DefaultKey)
	if err != nil {
		fmt.Println(err)
	} else {
		fmt.Println(data)
	}

}
```
2. Run it.
3. See error
```
Started rfid reader.
mfrc522: wrong number of bits 4
mfrc522: wrong number of bits 5
mfrc522: wrong number of bits 4
```

**Expected behavior**
```
Started rfid reader.
[18 100 14 52 76]
Write OK
[72 97 108 108 111 32 87 101 108 116 0 0 0 0 0 0]
```

**Platform (please complete the following information):**
 - OS: Raspbian GNU/Linux 10 (buster)
 - Board: Raspberry PI 3B+

**Additional context**
I read out the information of both tags, the only difference I see is the UID.

[image]
[image]
[image]
[image]

UdoWeberJR, 2021-04-07:
I changed the power supply and it worked, sorry for opening an issue.

closed 2021-04-07 by UdoWeberJR

maruel, 2021-04-07:
Wow thanks for the detailed report.

Two requests:

* I'd recommend to switch to v3, as improvements are only done there now. See https://periph.io/news/2020/a_new_start/
* Would you mind submit a PR to add documentation under https://periph.io/device/ for the mfrc522 as it seems to be a popular device and you could add a note on how to diagnose when there's insufficient power? It's a matter of adding a new file under https://github.com/periph/website/tree/master/site/content/device and an image under https://github.com/periph/website/tree/master/site/static/img.

UdoWeberJR, 2021-04-07:
Thanks for the hint, I changed to v3. 

I also added documentation as far as my knowledge goes.
Tried to upload a picture as well, but get message "Uploads are disabled. File uploads require push access to this repository."

To be honest, I do not know if it really was the power supply, just read somewhere on the web, that some hardware can be sensitive.  So it was just a guess, I changed it and it worked. So I have no way for diagnose.
But still, I added a hint in the documentation. 

[image]