google/periph #460: pmem example code incorrect

martin2250, 2021-05-05:
I think there is an error in the pmem example for MapAsPOD.

```go
package main

import (
	"log"

	"periph.io/x/periph/host/pmem"
)

func main() {
	// Let's say the CPU has 4 x 32 bits memory mapped registers at the address
	// 0xDEADBEEF.
	var reg *[4]uint32
	if err := pmem.MapAsPOD(0xDEADBEAF, reg); err != nil {
		log.Fatal(err)
	}
	// reg now points to physical memory.
}
```

It only works when I add a referencing operator to the function call, like so: `pmem.MapAsPOD(0xDEADBEAF, &reg)`, so this code works for me:

```go
var reg *[SMC_MEM_LEN / 2]uint16
if err := pmem.MapAsPOD(SMC_MEM_START, &reg); err != nil {
```

Otherwise, I get an error during runtime: `pmem: require Ptr to Ptr, got invalid` (or `got slice`/`got array`).

In case this is relevant, I'm compiling for an AT91SAM9G45 with Linux 3.0.0.

maruel, 2021-05-05:
Can you file the issue on github.com/periph/host instead? periph.io/x/periph is deprecated.

martin2250, 2021-05-12:
just did, thanks!

closed 2021-05-12 by martin2250